ExpressLRS/ExpressLRS #2059: Correct batter voltage is not sent back to the TX

miles2912, 2023-01-28:
## Current Behavior

Using a simple PWM RX (Radiomaster ER5A) and a 4.8v battery the telemetry data gives an incorrect voltage reading.  

## Steps to Reproduce

I used a different non-ELRS RX and it reads the correct voltage.   Verified with a voltage meter.   Another user on the Facebook ELRS group with an almost identical setup as me is getting the same results.

## Details
Very simple setup.   The Receiver is plugged into the battery and I used a 750mah 4.8v pack to power the receiver.  It binds just fine but under telemetry it gives an incorrect reading.   It says 4.0v but drops over a minute to 2.9v.  The actual pack is at 5.5v.

## Your Environment

- TX hardware:   TX16s  running edge 2.8 Ranger Micro - ELRS 3.2
- RX hardware:  ER5A on ELRS 3.2

- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):

CapnBry, 2023-01-31:
The Radiomaster ER5A samples I have received do not support VBAT telemetry, it isn't connected in hardware, so the reported value is just it randomly sampling the air. The BATT pins on the receiver are simply for providing power unless they've made a new revision.

closed 2023-02-04 by miles2912

pkendall64, 2023-02-04:
@miles2912 I believe the S pin is actually connected to the ADC.
You would of course have to connect it with a resistor divider network like in this image. The S pin is 1volt max input.
[image]

CapnBry, 2023-02-04:
Paul is right, it can be modified to report, as the trace is connected to the ADC, but with no divider on it. Does your schematic take into account the 220k resistor already on it, @pkendall64?

pkendall64, 2023-02-05:
> Paul is right, it can be modified to report, as the trace is connected to the ADC, but with no divider on it. Does your schematic take into account the 220k resistor already on it, @pkendall64?

Not sure how much that resistor would effect it would be an in series 220 ohms after the divider. But I could easily be wrong!