ExpressLRS/ExpressLRS #2061: BetaFPV AIO 5A 2022 don't go to binding mode when Plug and Unplug 3 times

kimman1, 2023-01-30:
## Current Behavior

Plug and Unplug battery 3 times but the FC didn't go to binding mode

## Steps to Reproduce
Just plug and unplug 3 times to go to binding mode like normal ELRS receiver

## Details
This fc: https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-5a-aio-brushless-flight-controller-elrs-2-4g
can't go to binding mode when plug and unplug 3 times. But binding phrase is OK

## Your Environment

- TX hardware: Jumper AION 2.4Ghz Nano TX
- RX hardware: The serial rx which is on the newest betafpv AIO FC
- Handset model: X9D+
- OpenTX version (including nightly number) : I'm using Edge TX in the latest version
- ExpressLRS version (TX & RX MUST MATCH): 3.2.0
- Packet Rate: 500Hz
- Telemetry Ratio: 1:16
- user_defines: Using default define

pkendall64, 2023-01-30:
Probably because they shipped it flashed with a bind-phrase! And when it is flashed with a bind phrase it will not go into traditional bind mode on 3 power ups.

closed 2023-01-30 by pkendall64

kimman1, 2023-01-31:
> Probably because they shipped it flashed with a bind-phrase! And when it is flashed with a bind phrase it will not go into traditional bind mode on 3 power ups.

Actually I've updated it the RX ( BETAFPV_AIO_RX target ) to the latest version and I make sure 100% don't have binding-phrase