ExpressLRS/ExpressLRS #2072: Tango 2 model ID not loading model ID on boot (breaks model match)

t413, 2023-02-09:
## Current Behavior

The Tango 2 (running current firmware version 1.3.5 / crossfire 6.19) always gives an expressLRS module a module ID == 0 and a _model mismatch_ error after boot. 

Switching models or changing the model ID seems to fix it sometimes. (but sometimes seems to crash the elrs firmware?!)

[image]

## Steps to Reproduce

1. Set up model match with an ID of 2, RX set to ID 2, connection works
2. Power cycle radio.
3. Radio won't connect to Rx.
4. Open ExpressLRS Lua script. Reports error _Model Mismatch_.
5. Change either the model or the model-match number.. it gets updated in ELRS and now works.

## Possible Solution (Not obligatory)

I'd say.. needs debugging on when the Tango2 sends this status?

## Your Environment

 always gives an expressLRS module a module ID of 0 and a _model mismatch_ error. 

- TX hardware: BetaFPV2g4micro
- RX hardware: skystars diversity
- Handset model: Tango 2 (running current firmware version 1.3.5 / crossfire 6.19)
- FreedomTX version: 1.3.5 dd3f9713
- ExpressLRS version 3.1.2 ISM2G4 4da3aa
- Packet Rate / Telemetry Ratio: see screenshots if needed

pkendall64, 2023-02-09:
Looks like a bug in the Tango 2 firmware. You should raise a bug against that instead. There was a problem with EdgeTX early on but they fixed it rather quickly.

t413, 2023-02-09:
Interesting @pkendall64, I'll see if I can find their issue / fix over in EdgeTX. If you had an easy way to find it there I'd appreciate the help finding it.

closed 2023-04-14 by github-actions[bot]