ExpressLRS/ExpressLRS #2075: Failsafe while within a few meters

neuhausj, 2023-02-11:
## Current Behavior

I had a strange failsafe with my glider this afternoon. 
It was the first flight session on ELRS 3.2 (I used 2.5.1 before).

I had been flying for 30min without issues at various distance and height, battery was still fine on both Tx and Rx (7.8-7.9v) and when the plane came back, at around 5m of me it announced “telemetry lost” and I saw the plane went into fail safe position (all servos). 
Luckily my brother manage to catch it before it would have crashed. 
This is extremely strange and worrying (plane is quite expensive). 

I tried to reconnect the radio again, but I didn't manage and the link was still shown as down.
I had to power cycle the Rx to go back to normal. I flew again 10min without issues.

I rarely saw less than -95dbm during the flights. LQ seemed to be ok too.

## Steps to Reproduce

I don't know

## Possible Solution (Not obligatory)

-

## Details

I have flown a lot with this plane in the past and it is the first time I faced this kind of issues. 
Here are the logs of my flights. I don't exactly know which flight was the problematic one, probably around 15h40-50.
[logs_Snipe.zip](https://github.com/ExpressLRS/ExpressLRS/files/10714495/logs_Snipe.zip)

## Your Environment

Middle of a field without any intereferences.

- TX hardware: HM Slim Pro Nano TX 
- RX hardware: Matek Vario
- Handset model: Tx16
- EdgeTX 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.2.0 on both with model match active
- Packet Rate: 150Hz (-112dbm)
- Telemetry Ratio: 1:8
- user_defines: 250mW dynamic power

macgyverek, 2023-02-12:
Today I hac\ve the same FS. My setup - 250mW fixed power, 250Hz, R9M +R9MM.  
I reconnected TX and was no link. After reconnect RX, link was back. It was about 1m from me.

macgyverek, 2023-02-21:
I have te same problem in v3.2. Its hard to reproduce but problem is still exist. 
My setup: Tx R9M and Rx R9MM LOCK_ON_FIRST_CONENCTION ON, Package Rate 200, telem 1:32, Hybrid, Tx Power 100 static,

Problem was after disarm when drone was about  10 cm front of antenna. After 5 second I hear "Telemetry Lost" and RED LED was flashing. V3 have a special algorithm and automaticaly decrease power if signal  is to strong. Before telemetry lost i saw 10mW on screen. After that I tried restart TX - nothing, restart RX - connection recovered. 

So I tried simulate that issue at home: 
1. Reflashed RX with LOCK_ON_FIRST_CONENCTION OFF and moving antenna from 10 cm to 1m. I have few connection lost but connection was established.
2. Reflashed RX with LOCK_ON_FIRST_CONENCTION ON and moving antenna from 10 cm to 1m. I reproduced that issue only once (I tried almost 1 hour). RED LED was flashing. After all, I change a rate to 25Hz and connection was recovered. So propably there some issue with power changes and LOCK_ON_FIRST_CONENCTION ON.

**Summary: Problem can be exist only with dynamic power and LOCK_ON_FIRST_CONENCTION ON.**

AlluteTech, 2023-03-20:
I have the same Failsafe within a few meters and unable to re-gain connection unless I power cycle the receiver.  I have ExpressLRS 3.2 with Pacakge Rate 200, tele 1:64, 250mW dynamic power, Frsky i6X with OpenText 1.9,, R9M2019 TX, and R9MX receiver. 

I think there two 2 parts of this issue. First, it failsafe on such a short distance and Second, it cannot reconnect back itself. 

I have spent days on troubleshooting this issue. First, I though it was a bad receiver, so I purchased another R9MX receiver, but the issue still persists. I then thinking this is antenna related, so I cracked opened my R9M super 8 antenna, check cables/wires, and there was no issue on the antenna for sure. Then I placed the antenna vertically instead of horizontally but issue persists. Finally, I move ExpressLRS back to 2.52 and I can run further this time, but it still failsafe at RSSI -120dBM. The problem is that I am unable to re-gain the RC link. The OSD would say NO RC LINK unable to arm.  I power cycle the TX and no help. I go to the model and check the receiver and I can see the receiver still has the red and green solid light but I am unable to regain the connection unless I power cycle the receiver.

Today, when I found this post, I re-flash the R9MX with the LOCK_ON_FIRST_CONENCTION OFF and change the Package Rate to 25Hz as mentioned by macgyverek, So far for a test run, I have not yet experience this issue. All connection was good and no failsafe. I hope this is a valid good workaround to this issue. But I will test more and report back.

macgyverek, 2023-03-20:
You can use 200Hz but with LOCK_ON_FIRST_CONENCTION OFF and static value of power. Now I flying without issue and FS.

AlluteTech, 2023-03-20:
Yes, I have set it OFF on ExpressLRS 3.2, and use static power and 200 package rate. I set this up in my long range FPV RC truck. However, during the 2nd test, it failsafe again, and I was unable to reconnect the radio link even though I walked very close to the truck. Eventually, I again have to manually powered cycle the receiver to get the RC link re-established. 

It seems that the LOCK_ON_FIRST_CONENCTION OFF and static power does not resolve my issue. 

Another test after that, I increased the power to 250mW and lowered the package rate to 25Hz, but the issue still persists. I started to wonder whether I am lucky enough to have two R9MX receiver defected? 

I have another R9Slim+OTA receiver that I use in another SCX-10 RC Truck for long range FPV, and so far it has no issue. I can go all the away to make it failsafe at RSSI -120dBm, but I can always reconnect the RC link back when I moved closer. I use Matek F411TE for my long range RC truck in order to see telemetry values on OSD.

AlluteTech, 2023-03-23:
I brought a brand new R9MX receiver and run 2 tests so far.  There is no failsafe within few meter problem for this receiver. I make the model go as far as it can in 10mW power and it did a failsafe at around -120dBm, but I was able to get closer and re-gain control without the need to power cycle the receiver. I am not sure why the previous 2 identical receivers kept having problem but this one did not. I will need to test more.

AlluteTech, 2023-03-25:
Unfortunately, after about 2 day and around 4 batteries, I have another failsafe just few meters in front of me when I was almost done with the testing and about to bring the model back. The RSSI is still good at around -80dBm at that time. In all my other failsafe condition, the RSSI normally reached -120dBm before it failsafe. So this -80dBm failsafe is unexpected. 

Plus when this type of unexpected failsafe occurred, I am unable to re-gain control of the model unless I power cycle the receiver. I mean in the other "normal" failsafe situation (RSSI > -120dBm), I am able to regain control by adjusting the antenna angle, or moving closer to the model, and / or increase the ExpressLRS TX power to the highest. 

However, I have not yet able to reproduce the issue on my another R9Slim+OTA receiver installed model. But I can be sure that I have tested 3 R9MX receiver with the same short distance failsafe issue so that could tell that the hardware may not be the issue. I have R9M and R9 receivers flashed with ExpressLRS 3.2. 

Middle of a field without any intereferences.

TX hardware: R9M2019
RX hardware: R9MX
Handset model: Flysky i6X
OpenText 1.9
ExpressLRS version (TX & RX MUST MATCH): 3.2.0 on both with model match active
Packet Rate: 200Hz
Telemetry Ratio: 1:32
user_defines: 250mW dynamic power

macgyverek, 2023-03-25:
Turn off dynamic power.

AlluteTech, 2023-03-25:
Thank you macgyverek. That is a good advise. I will test it this weekend with a fixed power (default 50mW or 250mW) and report back.

AlluteTech, 2023-03-27:
Just finished three 5200maH battery testing for this weekend. Total run time is around 2+ hours on FPV truck. I was not able to reproduce the Failsafe within a few meters after changing ExpressLRS to a Fixed power. I used to having this issue at least once or multiple times during one battery run. I tested a fixed power on 100mW and 250mW. 

Again the model never had any Failsafe in a few meter occurrence. I'd say this is Dynamic power Turn off would be the workaround solution for this issue.

Just notes that my another receiver R9Slim+OTA has no problem even with Dynamic power turn on; therefore, I should concludes that the Dynamic power is not compatible with R9MX.

closed 2023-07-10 by github-actions[bot]