ExpressLRS/ExpressLRS #2077: Incorrect Model ID Value Causing Unexpected Checkbox Selection (HappyModel 3.0.1)

edernucci, 2023-02-12:
**Description:**
**HappyModel** is unintentionally setting the model id value to 0 instead of the standard value of 255 used by other manufacturers. As a result, the javascript that interprets whether the checkbox should be selected or not does not understand the value 0 and leaves the checkbox selected without user interaction, causing the firmware update with the model match feature enabled.

**Steps to Reproduce:**

Set the model id value to 0
Observe the behavior of the checkbox in the javascript

**Expected Result:**
The checkbox should not be selected without user interaction.

**Actual Result:**
The checkbox is selected without user interaction.

**Impact:**
This issue can cause unintended firmware updates with the model match feature enabled, leading to potential compatibility issues.

**Solution:**
The company should change the model id value to the standard value of 255 to resolve this issue.

[image]
[image]

edernucci, 2023-02-12:
**Update:** After further testing, we have realized that this issue may also affect other manufacturers as long as they are on version 3.0.1, leading us to believe that the issue is in this version.

**Impact:**
This issue could potentially impact not just HappyModel but other manufacturers as well who are using version 3.0.1. This could cause unintended firmware updates with the model match feature enabled for a larger number of users, leading to compatibility issues.

[image]

iscarelli, 2023-02-12:
Exactly same behavior here.

3.0.1 has Model ID set as 0 and activating Model Match.

[]'s
Ismael

pkendall64, 2023-04-01:
I'm not really sure what the problem is here. I've flashed a device with 3.0.1, with a full erase and the default is that Model match is disabled. I can enable it and enter a value of 0 and save and it remembers that it is enabled with a 0 matched model. This is a valid configuration.
Perhaps happy model have done some testing and are shipping out with model match enabled?

closed 2023-04-14 by github-actions[bot]