ExpressLRS/ExpressLRS #2098: Configurator doesn't see devices in  home wifi network 192.168.AnotherBut_1.XXX

wsnzone, 2023-02-24:
I have home wifi network in 192.168.100.XXX subnetwork. Elrs configurator don't see my wifi devices.
But I can access it in browser, looking for address in my wifi router (clients list).
Could you add field to choose where to find elrs wifi devices? Or maybe configurator can pick in from current network computer ip address?

This behavior present on any elrs device. Radiomaster boxer (elrs internal), HM 2400 pro TX, EP recivers.
[image]
[image]

pkendall64, 2023-02-25:
If it's not detecting devices then most likely you have a problem with mDNS on you computer as that is what is used in the detection of devices.

wsnzone, 2023-02-25:
> If it's not detecting devices then most likely you have a problem with mDNS on you computer as that is what is used in the detection of devices.

I have the latest version of configurator? Do I need to install addition software on my computer? Or is it my router settings?

pkendall64, 2023-02-25:
The first answer in this stack overflow question should help
https://superuser.com/questions/491747/how-can-i-resolve-local-addresses-in-windows

wsnzone, 2023-02-25:
> The first answer in this stack overflow question should help https://superuser.com/questions/491747/how-can-i-resolve-local-addresses-in-windows

I have win10 build 19045.2364, it should work by default as I've read this post correct, isn't it?

pkendall64, 2023-02-25:
You can tell if mDNS is working correctly on your computer by installing an mDNS browser. If it is working then you should see all sorts of things on your home network.

miles2912, 2023-03-06:
Maybe tr turning on discovery on the local network?    https://www.asus.com/support/FAQ/1049382/#Win10

pkendall64, 2023-04-01:
@wsnzone Did you fix/diagnose your problem so we can close this issue?

wsnzone, 2023-04-01:
> Maybe tr turning on discovery on the local network? https://www.asus.com/support/FAQ/1049382/#Win10

It was turned on
[image]

wsnzone, 2023-04-01:
> 

I can't google mDns browser
I found this https://github.com/jldupont/mdns-browser
But there isn't any exe file to start it

wsnzone, 2023-04-01:
> @wsnzone Did you fix/diagnose your problem so we can close this issue?

I couldn't get it to work. I think main issue with my network addresses pool is not standart.
192.168.100.* not default 192.168.1.*. Could you send me the link to mDns browser for Windows?

pkendall64, 2023-04-01:
https://hobbyistsoftware.com/bonjourbrowser

Also, there is a field where you can enter the IP address manually, See the word "MANUAL" next the "WIFI DEVICES" on the screenshot toy put in. Clink on that and it will allow you to enter the IP address.

closed 2023-04-14 by github-actions[bot]