ExpressLRS/ExpressLRS #2108: Taranis X9D radio runs elrs at 400k baud just fine but at higher baud rates just prints errors

xgg69, 2023-03-07:
I have a BetaFPV Micro TX 2.4g and taranis x9d (not the new 2019 access version)

at 400k baud rate and 500hz refresh rate everything works fine but f1000hz says baud rate too low but when i increase baud rate to anything higher elrs module works fine but taranis constantly prints errors and lua script doesnt load

JyeSmith, 2023-03-07:
The x9 inverter isnt great and you can either enable onebit mode or do a hardware mod.  More info here. https://github.com/EdgeTX/edgetx/pull/59

xgg69, 2023-03-08:
I did already enable onebit mode but that didn't help so the hardware inverter mod is probably the way to go

kimman1, 2023-03-09:
> I did already enable onebit mode but that didn't help so the hardware inverter mod is probably the way to go

Actually I've done both of them select onebit sample and hardware mod ( replace R82 resistor ) but didn't work so I think just another way to change TX to a newer version or something like TX16s ( tested on 921k Baudrate ) 
I had found many solution and none of them work 
Do you think we need to replace the inverter mod like QX7 ?

xgg69, 2023-03-09:
Yeah i did both things too and still doesn't work

I am already planning to upgrade to a tx16s

I also wanna see if bypassing the inverter completely helps (if you can do that)

kimman1, 2023-03-10:
> Yeah i did both things too and still doesn't work
> 
> I am already planning to upgrade to a tx16s
> 
> I also wanna see if bypassing the inverter completely helps (if you can do that)

I think we can do it by replace the transistor which is responsible for invert the signal to the signal inverter chip like QX7 mod ( the inverter mod for crsf ). 
Do you thinking what I'm thinking ?

pkendall64, 2023-04-01:
I going to mark this issue as "closing" as it's not a firmware issue, but rather a hardware issue.

closed 2023-04-14 by github-actions[bot]

SkullerXfury, 2023-04-16:
> I think we can do it by replace the transistor which is responsible for invert the signal to the signal inverter chip like QX7 mod ( the inverter mod for crsf ). Do you thinking what I'm thinking ?

Do you have any progress on this bcz I am also having the same issue with my 2016 Taranis X9D plus the telemetry is losing and its take a lot of time to load the Lua in 400k baud rate

xgg69, 2023-04-26:
> Do you have any progress on this bcz I am also having the same issue with my 2016 Taranis X9D plus the telemetry is losing and its take a lot of time to load the Lua in 400k baud rate

i also run on 400k and my lua script loads fine, losing telemetry also happens but thats fine i have osd

kimman1, 2023-05-19:
Actually I've mod it with inverter chip and just only get maximum to 1.87Mbps

davidglt, 2024-03-10:
> Actually I've mod it with inverter chip and just only get maximum to 1.87Mbps

Hi kimman1,
Could you give us more information about the inverter chip mod in old X9D?
I've the same TX, after change de resistor R82 and check the onebit, I can work at 400k with betafpv elpr v3.3.2 module at 250hz without issues, telemetry on 1:64, edgeTX 2.9.4
Now I wanna try 500hz but 400k is not enough.
Cheers,
David

hfangit, 2024-08-09:
> Hi kimman1, Could you give us more information about the inverter chip mod in old X9D? I've the same TX, after change de resistor R82 and check the onebit, I can work at 400k with betafpv elpr [...]

I've moded my old X9D+, removed Q9 and R82, replaced with inverter chips such as SN74LVC1G04, SN74LV1T04 (That can work on 1.87Mbps)
[image]
[image]