ExpressLRS/ExpressLRS #2127: ELRS AP Mode servers in Class A IP Range and collides with corporate space

gusarov, 2023-03-18:
## Current Behavior

When I'm connecting to ELRS WiFi using USB wifi dongle on ethernet-conected PC, the ELRS collides both with my home (10.0.x.x) and corporate VPN (10.1.x.x) address spaces. Unable to open ELRS Page. Yes, I'm using 10.0.x.x for home.

## Steps to Reproduce

1. On a Ethernet connected PC with 10.0.x.x address, add a wifi module and connect to ELRS AP.
2. Open 10.0.0.1
3. Instead of ELRS, obviously I'm getting my home router page.

## Possible Solution (Not obligatory)

I suspect that the only reason for Class A Private address space for AP here is - it's simplicity to type in. But it is more important for me to type in my home networks smart devices address, then one time AP mode device... In fact this does not sounds as a good decision, because the whole point of ELRS AP is a quick temporary connection with one machine.

1) So, ideally it should use smallest possible class, which is Class C Address Space with a random subnet. E.g. 192.168.RND.1 (Likely RND >1 because very often home subnets are 0 or 1 in same range). This should be fine, because nobody usually have to type it in, it pops up configuration page automatically.

2) Alternatively it could be static Class B address (I never seen anybody uses Class B Private address at all) 172.16.0.1. A bit longer, but at least it will not collide with corporate networks.

3) Finally, there could be a config section to select predefined or custom IP range for AP mode. You probably raise an argument about "well known ELRS address", and backward compatibility. Fine, let's make it configurable at least for those who are suffering from this approach.

Every time when I need an upgrade I'm struggling, trying to switch different AP, connect from different machines, try to deal with network security when ELRS joins my network, and in most cases after lot's of frustration I'm just copying firmware to my phone and uploads through the phone, which is... you know... I don't want to buy another separate router for this... yet... Let's make things right and stop abusing corporate IP range.

pkendall64, 2023-03-20:
This is highly unlikely to happen as there are now so many users that are used to this, and it's all the youtuber videos and all the documentation.

gusarov, 2023-03-20:
> This is highly unlikely to happen as there are now so many users that are used to this, and it's all the youtuber videos and all the documentation.

That's why I suggested ```#4```, let's make it configurable, e.g. through custom defines

jawz101, 2023-05-08:
It was a poor design decision to use a [class A address](https://en.wikipedia.org/wiki/Private_network) for this sort of purpose.  Preexisting youtube videos shouldn't drive this.

brianmontantes, 2023-06-15:
@jawz101 @gusarov  It is very common for random devices, appliances and 3rd party VPN networks to use the default Class A (10.0.0.x) or C (192.168.1.x) address spaces. I'm not saying it is right, but in the real world gentlemen, it happens ALL THE TIME. I manage over 300 retail locations and I've seen some crazy stuff out there. For security, I always set these default networks as unroutable blackhole VLANs.

What I do is connect to an ELRS device at first using my cell phone, enter my home lab WiFi network details and just access/flash it when it becomes a client on my network. ExpressLRS Configurator even sees the client on my network and picks it up automatically as a flash target. Then when I'm out in the field and need to make changes, I can either setup a hot spot on my phone or just let it default back to 10.0.0.1 and connect to it on my phone. 

My takeaway: It is a poor decision to use 10.0.0.x as your default home or corporate subnet. Don't expect the real world to conform to what is "technically correct". Save yourself the heartache and just have a unique enough IP Addressing schema. Or just use Class B, no one ever uses Class B! ;-)

brianmontantes, 2023-06-15:
By the way, Class A 10.x.x.x isn't specifically designated as "Corporate" range. Stop calling it that, there is no such thing. It's all private, unroutable and can be used by anyone. Also, what "security" issues are you expecting when joining an ELRS device as a client? So you use a Class A range, but don't have the network equipment available to setup a vLAN or DMZ?

gusarov, 2023-06-19:
"It is a poor decision to use 10.0.0.x as your default home or corporate subnet."
@brianmontantes, it is an industry standard. All IT companies I've been working on in the last 20 years, including SNP500 companies uses Class-A private range that leading networking professionals configured. I can double check with folks from Google, Facebook & Microsoft if you have a doubts about their ranges, because I never had. I know that it is not limited to corporate usage, but it is obviously the primary use case, because of number of clients they had, intermediate routers, bridges, branches, etc...
The poor decision is to use Class-A for one-time temporary OTA access points like this... With 1 client only, and we already have people in this thread that supports this idea. I agree that in real world smart home devices often serves in CLASS C range, which is usually fine, but not for Class-A. ELRS project is the only thing I've ever seen that hosts one time configuration page at 10.0.0.1, do you another reasonable popular and certified example?
Now we have a drone department, and a big corporates have a huge requirements regarding LAN & WiFi security, sometimes including disabling SSID broadcasting at all. Projects like this spins up a tons of paperwork for review and approval by internal security committee + dedicated equipment and VLANs just for ELRS, just because someone decided that "typing 10.0.0.1 is faster and it is already on a youtube". And we was not meant to touch or develop communication part, we mean to develop [NDA] instead, but now sounds like this gap is not closed for community, either every company had to spend bunch of man-hours to security negotiation or to forking and patching anyway...

brianmontantes, 2023-06-19:
@gusarov Obviously you misunderstood. You used 10.0.0.1-254 as your default network, and now you see what a poor decision that was. Using the very first range in any IP Class is clearly not a good idea in the real world for a production network. Something you would know if you had some real world networking IP Schema experience. 

Anyone can use Class A, B or C. Just because of your poor planning, unique circumstances or inability to request a simple wireless vLAN sandbox, does not constitute a need for a change to the code base for everyone else. Just build the firmware from source with whatever IP address you want, problem solved. Have a good day.

closed 2023-06-19 by pkendall64

gusarov, 2023-06-19:
@pkendall64 why you are talking just about me? I'm trying to raise a problem to another level, this is a situation that every company have at workplace. Do you want me to create a new one with same title and another introduction? You just accused a network administrators of world top IT companies in making a poor decision and closed a bug, nice job!
And you still have not provided any example of legitimate device that behaves similarly...

pkendall64, 2023-06-19:
@gusarov  I did not say anything about any body. I just closed the ticket as I really do not think we will be changing this any time soon.