ExpressLRS/ExpressLRS #2133: sbus refractor

secretman28, 2023-03-20:
## Current Behavior
cant compile, setting the power(eu ism etc) un available on the configurator if i use the original github from the original coder all work #1

cant compile cannot choose the region power spec

## Steps to Reproduce
try to build the refractor for a rx

Warning! `src_filter` configuration option in section [env:native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:AXIS_THOR_2400_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:AXIS_THOR_2400_RX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Frsky_TX_R9M_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Frsky_TX_R9M_LITE_PRO_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Frsky_RX_R9MM_R9MINI_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Frsky_RX_R9SLIM_via_BetaflightPassthrough] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Frsky_RX_R9SLIMPLUS_via_BetaflightPassthrough] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Frsky_RX_R9MX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:Jumper_RX_R900MINI_via_BetaflightPassthrough] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:HappyModel_TX_ES915TX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:HappyModel_TX_ES900TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:HappyModel_RX_ES915RX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:HappyModel_ES24TX_2400_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:HappyModel_ES24TX_Pro_Series_2400_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:BETAFPV_900_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:BETAFPV_900_TX_MICRO_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:BETAFPV_900_RX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:BETAFPV_2400_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:BETAFPV_2400_TX_MICRO_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:BETAFPV_2400_RX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:GHOST_2400_TX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:GHOST_ATTO_2400_RX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_VOYAGER_900_TX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_VOYAGER_900_OLED_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_VOYAGER_900_RX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_VOYAGER_900_ESP_RX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_FLASH_2400_TX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_FLASH_2400_OLED_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_FLASH_2400_RX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:NamimnoRC_FLASH_2400_ESP_RX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:FM30_TX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:FM30_RX_MINI_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:FM30_RX_MINI_AS_TX_via_STLINK] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:HGLRC_Hermes_2400_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:QuadKopters_JR_2400_TX_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_TX_TTGO_V1_SX127x_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_TX_TTGO_V2_SX127x_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_TX_ESP32_SX127x_E19_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_TX_ESP32_SX127x_RFM95_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_RX_ESP8285_SX127x_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_RX_HUZZAH_RFM95W_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_900_RX_PWMP_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DIY_2400_TX_ESP32_SX1280_Mini_via_UART] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
[... cut ...]
Compiling .pio\build\HappyModel_EP_2400_RX_via_BetaflightPassthrough\lib598\ESP8266mDNS\ESP8266mDNS.cpp.o
========================= [FAILED] Took 12.42 seconds =========================

Environment                                      Status    Duration
-----------------------------------------------  --------  ------------
HappyModel_EP_2400_RX_via_BetaflightPassthrough  FAILED    00:00:12.416
==================== 1 failed, 0 succeeded in 00:00:12.416 ====================

add the setting of region power available in the configurator 

## Your Environment
win10

pkendall64, 2023-03-20:
The branch/PR has been merged into master, just choose the master branch from configurator.

pkendall64, 2023-03-20:
If you need help with this, please ask on discord rather than opening a ticket.

closed 2023-03-20 by pkendall64