ExpressLRS/ExpressLRS #2136: BLE Joystick, CH8 not mapped to axis.

ksmigrod, 2023-03-24:
With new BLE Joystick code (master CE_LBT 29d581) my Radiomaster Zorro (EdgeTX v2.8.1) does not send CH8 to computer.

I've tried mapping S2 slider and SC three way switch to CH8. CH8 is displayed correctly on Zorro's builtin display, but it is "0" on Bluetooth. I've tried Linux (with Intel Corp. AX210 Bluetooth), Windows 11 (with Asus USB BT-500) and an Android 13 phone. On each device axis CH1 to CH7 work, and buttons CH9 - CH12 also work, but CH8 does not.
The same switches and sliders work when moved to CH5 - CH7.

pkendall64, 2023-04-01:
With 3.x.x-maintenance (upcoming 3.2.1) I set up a model in EdgeTX with channels 1-4 on the sticks, channels 5-8 on switches.
Connected via BLE to Mac, went into Liftoff and calibrated the sticks, moved to the buttons section and I could change the reset action to any of the four assigned switches. So this does not appear to be a problem with ELRS BLE code.

closed 2023-04-14 by github-actions[bot]

mmilata, 2026-01-04:
However I can see the number change in `hid-recorder` - S2 changes the second "Slider":
```
# ReportID: 3 / Button: 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
#             | X:  16393 | Y:  16393 | Z:     19 | Rz:     19 | Rx:     19 | Ry:  16393 | Slider:  16613 ,  16393 
#             | Hat switch:    0 
E: 000033.134817 20 03 00 00 09 40 09 40 13 00 13 00 13 00 09 40 e5 40 09 40 00
```
It's a little bit suspicious that both have the same usage in HID descriptor (0x36 Slider), perhaps this confuses linux? https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/BleGamepad.cpp#L424-L432

EdgeTX USB has "Dial" usage for the last axis:
```
#  Button: 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 | X:   1024 | Y:   1023 | Z:      0 | Rx:   1024 | Ry:      0 | Rz:      0 | Slider:   1035 | Dial:   1006 | Battery Strength:  100 
E: 000001.684807 20 00 00 00 00 04 ff 03 00 00 00 04 00 00 00 00 0b 04 ee 03 64
```

Gonna try changing `0x36` to `0x37` [here](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/BleGamepad.cpp#L432) once I figure how to build this.

mmilata, 2026-01-07:
MacOS 26.1:
- on 3.6.2 I can't see CH8
- with the patch I can

Only tested on https://hardwaretester.com/gamepad