ExpressLRS/ExpressLRS #2139: BETAFPV SuperD ELRS 2.4G RX wrong diversity behavior

StiOII, 2023-03-25:
## Current Behavior

I bought two SuperD ELRS 2.4G receivers, and both of them demonstrate the same issue. One of the antenna's RSSI is never changing, active antenna number is never changing, even when I hold antennas in hand.

It looks like this receiver doesn't work as diversity but as a single channel receiver. This is not an expected behavior, and since both of them have the same issue, I believe, it's not a HW problem.

## Steps to Reproduce

Just normal use

## Your Environment

- TX hardware: Happymodel ES24TX Slim Pro 2.4G
- RX hardware: BETAFPV SuperD ELRS 2.4G
- Handset model: Taranis x9 lite
- OpenTX version: 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): 3.2.0 (latest script too)
- Packet Rate: different
- Telemetry Ratio: ???
- user_defines: see screens

--

https://youtu.be/5R9MbetvL8U

[image]
[image]

pkendall64, 2023-03-25:
It's not a bug, it's just a behaviour of the current implementation. This is also related to #2109.
Both radios receive the packet and the firmware reads the first one sets the RX_DONE flag, if they both do it then it will generally favour one side. We only read the second side if the first side fails it's CRC check, and we only read the RSSI/SNR from the side that passes its check.
There is a branch that is being worked on to "fix" this behaviour, but it is in the testing phase and not ready for general consumption. If you would like to test it, the branch is `true_diversity_rssi_update`.

StiOII, 2023-03-25:
Thank you very much Paul, for the explanation!

I also have Matek diversity RX and as far as I remember, it has a different behavior. That's why I was confused.
Anyway it would be nice to see both RSSIs of course.

I will try the branch you've mentioned.

Can be closed.

pkendall64, 2023-03-25:
Yes, the antenna diversity has a different behaviour, and that's what we are trying to emulate with the branch, so it looks consistent for users.

We will leave the ticket open until a pull-request is made from the branch and merged, as it helps with tracking.

Garwol4130, 2023-04-01:
> It's not a bug, it's just a behaviour of the current implementation. This is also related to #2109. Both radios receive the packet and the firmware reads the first one sets the RX_DONE flag, if they [...]

"fix" ? not a bug? In my case receiver (tested on both  SuperD and EP1 dual)for some reason decides that active radio is always the second one, and rssi that is updating is 2RSS and I can't put that in osd so I don't know my RSSI value when flying.. this is clearly a bug...

pkendall64, 2023-04-01:
> "fix" ? not a bug? In my case receiver (tested on both SuperD and EP1 dual)for some reason decides that active radio is always the second one, and rssi that is updating is 2RSS and I can't put that [...]

Even if we make adjustments to the way we collect RSSI for both radios, you will continue to have the issue in your OSD as you can only display 1RSS as you say. So I guess you'd call that a bug in your FC firmware.

StiOII, 2023-04-08:
I've tried true_diversity_rssi_update - looks very nice! At lease I can see in logs what happens with both antennas in flight.

closed 2023-04-19 by JyeSmith