ExpressLRS/ExpressLRS #2148: ELRS-RP3 3.2 updated as MATEK R24-D 2.4GHz RX

NikosSprocket, 2023-04-01:
## Current Behavior

Updated a RadioMaster RP3 Rx from 3.0 to 3.2.0 using Configurator v1.5.5 with the following:

- Device category: RadioMaster 2.4 GHz
- Device: RadioMaster RP3 Diversity 2400 RX

The compiler starts with "Processing MATEK_2400_RX_R24D_via_WIFI"; log is attached.
After a successful flash, I inspected the infromation and I got the following:

- WiFi interface of the Rx displays: MATEK R24-D 2.4GHz RX, Firmware Rev. 3.2.0 (8d3388) ISM2G4
- TeleMaster TX16S running ELRS 3.2.0 ISM2G4 8d3388: Other Devices -> RM RP3 2400RX

Telemetry is working but I have not flown since the update to 3.2.0.

**_IMPORTANT_**: is it safe to fly with this configuration?

- TX hardware: TX16S MKII
- RX hardware: RP3
- EdgeTX: v2.8.1
- ExpressLRS version (TX & RX MUST MATCH): 3.2.0
[ExpressLRSBuildLog_20230401025932423.txt](https://github.com/ExpressLRS/ExpressLRS/files/11128362/ExpressLRSBuildLog_20230401025932423.txt)
[image]
[image]

pkendall64, 2023-04-01:
These targets are exactly the same hardware layout. It is perfectly safe to fly.
This is because the RP3 is an alias for the Matek. These kinds of inconsistencies are being worked through.

closed 2023-04-01 by pkendall64

NikosSprocket, 2023-04-01:
Thank you Paul for the super fast response!

I had that suspicion but I just wanted to make sure before I had an unscheduled failsafe and RTL (Arducopter)!

Many, many thanks for an awesome system!

Cheers!
Nikos

x4code, 2023-04-03:
already closed? fixed?

NikosSprocket, 2023-04-04:
I had no problems. Just making sure the firmware compiled was the correct for RP3. It sounds like MATEK

pkendall64, 2023-04-04:
All good. This problem will be solved when we move to cloudbuilds in Configurator.