ExpressLRS/ExpressLRS #2152: RadioMaster RP3 Diversity 2.4GHz RX Wrong Name after Flash

x4code, 2023-04-02:
## Current Behavior

After Flashing RadioMaster RP3 Diversity 2400 RX 3.2 ELRS shows as MATEK R24-D 2.4GHz RX on webpage and build seems to be Matek.

## Steps to Reproduce
Choose this options:
3.2
RadioMaster 2.4 GHz
RadioMaster RP3 Diversity 2400 RX

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

when i received it was labed on rx webpage as : (Correct)
RadioMaster RP3 Diversity 2.4GHz RX
Firmware Rev. 3.1.2 (4da3aa) CE_LBT

After Flashing to 3.2 now shows: 
MATEK R24-D 2.4GHz RX
Firmware Rev. 3.2.0 (8d3388) ISM2G4

## Your Environment

- TX hardware:
- RX hardware: RadioMaster RP3 Diversity 2.4GHz RX
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):  3.1.2  to 3.2
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2023-04-02:
This is a duplicate of #2148 
This will be fixed in a future release.

pkendall64, 2023-04-02:
No worries.

closed 2023-04-02 by pkendall64