ExpressLRS/ExpressLRS #2159: Radio connectivity when quad is powered on first

socalengineer, 2023-04-05:
With version 2.5.2 it doesn't matter if I power on the quad or the radio first, it always connects and establishes a link.

With 3.2.0 I have to always power on the radio. Is this the intended behavior? It seems to be the behavior in 2.5.2 is much better.

pkendall64, 2023-04-05:
You have not provided enough information, like hardware etc. Have you asked on discord, you would get much faster help asking there.

socalengineer, 2023-04-05:
BetaFPV Nano TX Module, and the Happymodel EP1 Receiver. It doesn't do this with the earlier version, 2.5.2 on the same hardware.

pkendall64, 2023-04-06:
Well I just tested this, powering on RX first, then powering on the Radio and it works as expected.
It can take a few seconds (up to 15 seconds) to get a connection as there are quite a few RF modes for the RX to check for a connection on.

socalengineer, 2023-04-06:
I will try it again and check thank you for your help.

CapnBry, 2023-05-09:
It takes longer on 3.x than it did on 2.x to cycle through all the modes, so if you plug your quad in first then turn on the radio, you're just gonna have to wait for it to circle around again (~15s as Paul mentioned). Obviously it may go to wifi before it connects if you power the quad so early that it hits that deadline first, but obvs you can change that wifi configurator.

socalengineer, 2023-05-09:
Yes that's what happens to me it goes to wifi because I power on the quad very early to get a satellite signal lock. How do I change that wifi configurator? It's in the build settings of expresslrs?

CapnBry, 2023-05-09:
[image]

But you probably don't want to set it up to be like 10 minutes just to wait for a GPS lock. Power the quad so it starts working on the GPS, turn on the transmitter, wait for them to connect, then turn off the transmitter. The receiver will lock on the signal and then not go to wifi mode at all after that. Bonus, when the transmitter turns back on, it will connect as soon as EdgeTX enables the radio, as the RX and TX are now locked on the same rate.

socalengineer, 2023-05-09:
Got it, thank you!

closed 2023-05-23 by github-actions[bot]