ExpressLRS/ExpressLRS #2164: Connect Home Wifi does not work

jasc76, 2023-04-09:
## Current Behavior

wifi connect to home network

## Steps to Reproduce
Restart RX

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details
RX: Happymodel EP1 Dual TXCO

If I set home wifi in Build the RX does not connect to home wifi. Password checke multiple times and did copy/paste. It is correct. SSID is seen in RX web site. If I hit connect through web page it shows error pop-up after a while.
Restart, tried again->same
Clicking forget home wifi did not work, then it worked. Then it didnt work
Trying one time also does not work
After  forgetting home wifi through web apge worked I tried to connect to my standard wifi, my second wifi all does not connect

My TX connects just fine !

## Your Environment

- TX hardware: AION nano
- RX hardware: HappyM Ep1 Dual TXCO
- Handset model: X20s
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 3.2.1
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2023-04-10:
The antenna for wifi on these devices is very small and needs to be fairly close to you wifi router for it to connect.

closed 2023-04-10 by pkendall64

jasc76, 2023-04-10:
> The antenna for wifi on these devices is very small and needs to be fairly close to you wifi router for it to connect.

I got full signal on the PC right next to the rx?! Thats not enough?
Also the TX connects just fine
And the RX sees the ssid in the webinterface.

jasc76, 2023-04-10:
wow, yes really short range