ExpressLRS/ExpressLRS #2188: Ghost TX Stlink flashing issue

jurgelenas, 2023-04-30:
I get this error when trying to flash Ghost TX through Stlink when using latest configurator nightlies with cloudbuilds support:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code

    exec(code, run_globals)
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/ae3406d8bdeadeccb3dc44ac0fe3c179f356eec9/firmware/flasher.pyz/__main__.py", line 3, in <module>
    if __name__ == '__main__':
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/ae3406d8bdeadeccb3dc44ac0fe3c179f356eec9/firmware/flasher.pyz/_bootstrap/__init__.py", line 253, in bootstrap
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/ae3406d8bdeadeccb3dc44ac0fe3c179f356eec9/firmware/flasher.pyz/_bootstrap/__init__.py", line 38, in run
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_configurator.py", line 389, in main
    return binary_flash.upload(options, args)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_flash.py", line 163, in upload
    return upload_stm32_stlink(args, options)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_flash.py", line 52, in upload_stm32_stlink
    stlink.on_upload([args.file.name], None, {'UPLOAD_FLAGS': [f'BOOTLOADER={options.bootloader}', f'VECT_OFFSET={options.offset}']})
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/stlink.py", line 59, in on_upload
    BL_CMD, APP_CMD = get_commands(env, firmware_path)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/stlink.py", line 28, in get_commands
    env_dir = env['PROJECT_PACKAGES_DIR']
KeyError: 'PROJECT_PACKAGES_DIR'
```

jurgelenas, 2023-04-30:
Same issue can be reported on HappyModel PP RX:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code

    exec(code, run_globals)
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/ae3406d8bdeadeccb3dc44ac0fe3c179f356eec9/firmware/flasher.pyz/__main__.py", line 3, in <module>
    if __name__ == '__main__':
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/ae3406d8bdeadeccb3dc44ac0fe3c179f356eec9/firmware/flasher.pyz/_bootstrap/__init__.py", line 253, in bootstrap
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/ae3406d8bdeadeccb3dc44ac0fe3c179f356eec9/firmware/flasher.pyz/_bootstrap/__init__.py", line 38, in run
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_configurator.py", line 389, in main
    return binary_flash.upload(options, args)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_flash.py", line 152, in upload
    return upload_stm32_stlink(args, options)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_flash.py", line 52, in upload_stm32_stlink
    stlink.on_upload([args.file.name], None, {'UPLOAD_FLAGS': [f'BOOTLOADER={options.bootloader}', f'VECT_OFFSET={options.offset}']})
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/stlink.py", line 59, in on_upload
    BL_CMD, APP_CMD = get_commands(env, firmware_path)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/stlink.py", line 28, in get_commands
    env_dir = env['PROJECT_PACKAGES_DIR']
KeyError: 'PROJECT_PACKAGES_DIR'
```

jurgelenas, 2023-04-30:
It looks like it is more generic issue with stlink flashing method in general:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,

  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/8dbbdbe442660713ba1de07828ceb20cdaa5e26a/firmware/flasher.pyz/__main__.py", line 3, in <module>
    if __name__ == '__main__':
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/8dbbdbe442660713ba1de07828ceb20cdaa5e26a/firmware/flasher.pyz/_bootstrap/__init__.py", line 253, in bootstrap
  File "/home/julius/.config/ExpressLRS Configurator Nightly/firmwares/cloud/ExpressLRS/8dbbdbe442660713ba1de07828ceb20cdaa5e26a/firmware/flasher.pyz/_bootstrap/__init__.py", line 38, in run
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_configurator.py", line 389, in main
    return binary_flash.upload(options, args)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_flash.py", line 163, in upload
    return upload_stm32_stlink(args, options)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/binary_flash.py", line 52, in upload_stm32_stlink
    stlink.on_upload([args.file.name], None, {'UPLOAD_FLAGS': [f'BOOTLOADER={options.bootloader}', f'VECT_OFFSET={options.offset}']})
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/stlink.py", line 59, in on_upload
    BL_CMD, APP_CMD = get_commands(env, firmware_path)
  File "/home/julius/.shiv/flasher.pyz_97275f2ed0e39b20fbb04f05c86aad84d08e777c51bbff37fdf549209c6340a5/site-packages/stlink.py", line 28, in get_commands
    env_dir = env['PROJECT_PACKAGES_DIR']
KeyError: 'PROJECT_PACKAGES_DIR'
```

pkendall64, 2023-04-30:
Ok, this is because the current stlink flashing method uses stlink executable included in platofrm IO 😭

PR #2197 (2023-05-08): Add pystlink/pyusb to binary flasher
These are added so there are no dependencies on external tools for the binary flasher.

Fixes #2188
closed 2023-05-08 by JyeSmith