ExpressLRS/ExpressLRS #2194: argument --flash: invalid UploadMethod value: 'dfu'

jurgelenas, 2023-04-30:
`dfu` flashing method found in the [SIYI hardware](https://github.com/ExpressLRS/ExpressLRS/blob/a626d4368f76d7df88e70abe2c9e67cd6f0a9e79/src/hardware/targets.json#L1518) is not recognized as a valid UploadMethod value:

```
usage: flasher.pyz [-h] [--dir DIR] [--phrase PHRASE] [--ssid SSID]
                   [--password PASSWORD] [--auto-wifi AUTO_WIFI]
                   [--no-auto-wifi] [--airport-baud [AIRPORT_BAUD]]
                   [--rx-baud [RX_BAUD]] [--lock-on-first-connection]
                   [--no-lock-on-first-connection] [--tlm-report [TLM_REPORT]]
                   [--fan-min-runtime [FAN_MIN_RUNTIME]] [--uart-inverted]
                   [--no-uart-inverted] [--unlock-higher-power]
                   [--no-unlock-higher-power]
                   [--buzzer-mode {quiet,one-beep,beep-tune,default-tune,custom-tune}]
                   [--buzzer-melody BUZZER_MELODY]
                   [--domain {eu_433,au_433,in_866,eu_868,au_915,fcc_915}]
                   [--target TARGET] [--flash {wifi,uart,bf,etx,stlink,dir}]
                   [--out OUT] [--port PORT] [--baud BAUD] [--force]
                   [--confirm] [--tx] [--lbt]
                   [file]
flasher.pyz: error: argument --flash: invalid UploadMethod value: 'dfu'
```

pkendall64, 2023-04-30:
Do we really care about this one?? I can add it if required, but it's just for a single target and it can still be flashed via platformio.

jurgelenas, 2023-05-01:
I do not care about it personally, just reporting an issue. We could also remove `dfu` flashing method from `targets.json`.

In general, if we have this device & flashing method in the configurator it should work.

CapnBry, 2023-05-09:
It is the only way to flash SIYI TX devices, which people use. Our PIO build supports it, what's .... what ... what is flasher.pyz?

jurgelenas, 2023-05-09:
> That's binary_flash.py? Can't we just shell out like we do now, but obvs we need more than python files to use the dfu-util binary

Yes it is binary_flash.py. If it is just a simple code change that is required to fix it I am all for fixing it before we release v3.3.

pkendall64, 2023-05-09:
It will be bringing in a new python library for DFU support. It's not a massive library but I can't test it.
https://github.com/square/pyfu-usb is the library I was thinking to use.

closed 2024-07-30 by github-actions[bot]