ExpressLRS/ExpressLRS #2212: Upgrade from 3.2.0 to 3.2.1 results in no-binding for Target "Flywoo EL24P 2400 RX"

Quas7, 2023-05-09:
## Current Behavior

Upgrading to 3.2.1 results in some kind of exception-loop.
No binding possible but web-interface is still working.

## Steps to Reproduce

Update GOKU F405 AIO with ELRS2.4G RX to 3.2.1
```
      target: 'DIY_2400_RX_ESP8285_SX1280_via_BetaflightPassthrough',
      source: 'GitRelease',
      gitTag: '3.2.1',
```

Logfile including 3.2.1 and 3.2.0 traces: [expressslrs-configurator_3.2.1_and_3.2.0_update_session.log.zip](https://github.com/ExpressLRS/ExpressLRS/files/11435838/expressslrs-configurator_3.2.1_and_3.2.0_update_session.log.zip)

```
      '--------------- CUT HERE FOR EXCEPTION DECODER ---------------\n' +
      '\r\n' +
      'Exception (28):\r\n' +
      'epc1=0x4010046c epc2=0x00000000 epc3=0x00000000 excvaddr=0x065d0000 depc=0x00000000\r\n' +
      '\r\n' +
      '>>>stack>>>\r\n' +
      '\r\n' +
      'ctx: cont\r\n' +
      'sp: 3ffffbc0 end: 3fffffc0 offset: 0190\r\n' +
      '3ffffd50:  0000000b 00000001 3ffffd30 f'
```

## Possible Solution (Not obligatory)

Stay on 3.2.0 as it just works fine.

## Details

If anybody wants to dive into this topic I would offer to test the code.

## Your Environment

- TX hardware:  Radiomaster TX12MK2 / Zorro
- RX hardware: Flywoo F405 AIO with ELRS2.4G
- EdgeTx version: 2.8.1
- ExpressLRS version (TX & RX MUST MATCH): 3.2.x
- Packet Rate: 500Hz
- Telemetry Ratio: 128
- user_defines: EU_CE, binding phrase, lock-on first, Auto-wifi, home-wifi+pass

pkendall64, 2023-06-01:
Since you are getting an exception trace printed on the serial connection you will probably need to reflash the device using the web-flasher https://pkendall64.github.io/elrs-web-flasher/ and on the last page choose the option to "erase flash" before clicking the flash button.

Quas7, 2023-06-09:
Thanks for hinting the web-flasher erase flash feature 👍

closed 2023-06-09 by Quas7