ExpressLRS/ExpressLRS #2214: No build target for BetaFPV ELRS Micro Receiver

RobertoD91, 2023-05-11:
There is no target for "Betafpv ELRS Micro Receiver" (official link: https://betafpv.com/collections/expresslrs-series-accessories/products/elrs-micro-receiver ).

BetaFPV's documentation is outdated and only covers ExpressLRS v2.

The targets "BETAFPV Nano 2400 RX" and "BETAFPV PWM 2400 RX" don't work.

The targets "DIY 2400 RX PWMP" seems to work, tested only with PWM servos

RobertoD91, 2023-05-11:
Maybe I should have opened the issue on ExpressLRS/ExpressLRS-Configurator.

If I understand correctly, `BETAFPV PWM 2400 RX` is an alias for `DIY 2400 RX PWMP`

pkendall64, 2023-05-12:
`BETAFPV PWM 2400 RX` is the correct target. So we'll close this ticket.

closed 2023-05-12 by pkendall64