ExpressLRS/ExpressLRS #2217: Configurator fails to build "HDZero_RX5.1_ESP32_Backpack_via_WIFI"

SkylarTheGrey, 2023-05-13:
## Current Behavior

Configurator fails to build "HDZero_RX5.1_ESP32_Backpack_via_WIFI" due to below error code.

`Error: Invalid environment name 'HDZero_RX5.1_ESP32_Backpack_via_WIFI'. The name can contain alphanumeric, underscore, and hyphen characters (a-z, 0-9, -, _)`

## Steps to Reproduce

1.Build backpack firmware on any release under the VRX device category and HDZero RX5.1 VRX Backpack device in configurator v1.5.9

2. Get Error Code

3. ???

4. Cry

## Possible Solution (Not obligatory)

 Should just need updated target name? "HDZero_RX5-1_ESP32_Backpack_via_WIFI" to remove the offending period?

## Details

Found while trying to build VTX backpack firmware with bind phrase. WiFi config page does not yet have that as a runtime setting for the FW level on goggles group 2.

`From here you can update your VRx Backpack module with HDZero_Goggle_RX5.1_ESP32_Backpack firmware

Firmware Rev. merge (472ae1) `

## Your Environment

any release 1.0, 1.1, 1.2
VRX device category
HDZero RX5.1 VRX Backpack device
configurator v1.5.9

pkendall64, 2023-05-13:
This has been fix in the master branch of backpack. Just flash the branch from configurator.
This is because platformio has made a change!

closed 2023-05-13 by pkendall64

baconninja87, 2023-05-24:
I am helping someone out with this issue during their flashing of a backpack for HDZ and although I have the master branch selected, there is no target for the HDZ backpack listed anymore.  What am I missing?

[image]