ExpressLRS/ExpressLRS #2234: ExpressLRS Joystick bluetooth cannot be used for simulators on PC for jumper t pro (and v2)

gumileeev, 2023-06-01:
I go to the script express lrs and activate the BLE Joystick. Then I find the device in the computer and connect it. But no simulator sees it. How to connect to the simulator via bluetooth? This function is not implemented for this?
I'm looking to use bluetooth to connect. No need to use usb cable.
Jumper t pro
3.2.1
PC: win10
sim: Liftoff Micro Drones

pkendall64, 2023-06-01:
Yes it is supported to use the joystick mode in simulators. I have Liftoff and it works perfectly there.
You can use something like https://gamepad-tester.com/ to see if it is detected and if the controls are moving as expected.

jawz101, 2023-06-13:
I haven't been able to get my Radiomaster Tx16s Mk 2 to work on [DCL - The Game on Steam](https://store.steampowered.com/app/964570/DCL__The_Game/).  Steam doesn't recognize the controller & the game immediately exits after the intro

It works on [Tiny Whoop Go](https://tinywhoopgo.com/), however

pkendall64, 2023-06-13:
> I haven't been able to get my Radiomaster Tx16s Mk 2 to work on [DCL - The Game on Steam](https://store.steampowered.com/app/964570/DCL__The_Game/). Steam doesn't recognize the controller & the game immediately exits after the intro
> 
> It works on [Tiny Whoop Go](https://tinywhoopgo.com/), however

Sounds like you need to raise an issue with DCL then. It works on Liftoff, Tryp, Velocidrone, CurryKitten Sim, FPV.Skydive and others