ExpressLRS/ExpressLRS #2249: AirPort connection connected but no data gets through

ScalyDragon, 2023-06-16:
## Current Behavior

AirPort serial connection not working

## Steps to Reproduce

1. Both (TX & RX) updated to 3.3.0-RC1

2. Paired using binding phrase

3. Activated AirPort functionality via WiFi on TX and RX

4. Set baudrate to 19200 and packet rate to F1000Hz

5. Connected serial TX and serial RX on ELRS RX side to each other (creating a loop)

6. Connected to TX via screen on Linux with baudrate 19200

7. Sending values with hterm or screen on baudrate 19200 has no effect, nothing comes back. Also no written value can be seen on oscilloscope between the RX-side serial connection, in fact nothing observable happens.

## Details

I want to establish a serial connection between a drone and EGSE

No

No connection loss was seen at any time and the USB-Serial-Interface was recognized correctly

## Your Environment

- TX hardware: BETAFPV TX micro 2.4GHz 1000mW
- RX hardware: BETAFPV RX lite 2.4GHz
- Handset model: Radiomaster TX16s (deactivated, TX-module used standalone with USB-cable attached)
- OpenTX version (including nightly number) EdgeTX 2.7.1 (a55aff02)
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0-RC1
- Packet Rate: F1000
- Telemetry Ratio: 1:2
- user_defines: default

pkendall64, 2023-06-17:
When using a TX micro you need to leave the switches in the flashing position and go to the hardware.html page and change serial RX to 3 and serial TX to 1 and remove the entries for backpack.

ScalyDragon, 2023-06-19:
Thank you very much, this solves my problem!

closed 2023-06-19 by ScalyDragon

zhaolei, 2024-08-26:
> When using a TX micro you need to leave the switches in the flashing position so the module can communicate over the USB port.

How to set in TX?