ExpressLRS/ExpressLRS #2254: 3.3.0 RC1 - AUX1 (CH5)  it remains stuck at 999 (R9M and R9Mini ttarget)

robustini, 2023-06-17:
## Current Behavior

With ELRS V3.2.1 I have always used AUX1 (CH5) with Ardupilot as a channel to activate some useful options.
So with a two-position switch, the value on the FC channel monitor varies from 999 to 2000.
I decide to try 3.3.0 RC1 and now it remains stuck at 999.
Downgrade to 3.2.1, identical settings, and it works again.

## Steps to Reproduce

1. Install 3.3.0 RC1

2. Go to the FC channel monitor, CH5 does not react to any change imparted by TX

3. Downgrade to 3.2.1 and all work fine

## Possible Solution (Not obligatory)

## Details

I have had others run the same test and it works for them but with different targets.

## Your Environment

- TX hardware: FrSky R9M
- RX hardware: FrSky R9 Mini
- Handset model:
- Edge TX V2.8
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0 RC1
- Packet Rate: 100 HZ Full
- Telemetry Ratio: 1:2
- user_defines:

CapnBry, 2023-06-17:
Confirmed. TL;DR, here is the fix for people experiencing it.
* Download this firmware [r9m_clearht.zip](https://github.com/ExpressLRS/ExpressLRS/files/11779784/r9m_clearht.zip)
, unzip the firmware file from it, and flash the R9M module with it.
* Load the Lua after flashing and verify the version number (bottom of the list) says `3.x.x-maintenance FCC915 4ab160`
* Flash whatever firmware you want, 3.3.0-RC1 or 3.2.1, the issue will be resolved.

### Details
Took me a while to be able to get a debug build compiled but what this is that HeadTracking is turned on. The config is telling it to override channels 5,6,7 with head tracking data. The problem is that the config structure has some garbage in it from the 2.x days, which gets carried along up to current. After the module has already been updated to 3.x the damage is already done and can no longer be detected. Not having a backpack, there's also no way to turn off the feature so you're stuck. The only way to fix it is to reset the config (by going back to 2.5.2 then upgrading again).

robustini, 2023-06-18:
All clear where the problem interspersed, thank you very much @CapnBry!

PR #2255 (2023-06-19): Fix no backpack overwriting channels
This fix basically disables head-tracking if a backpack is not detected, i.e. backpack has not responded to the version query.
For head tracking the backpack would have to be updated for that feature and that also includes the version query response code.

As mentioned in the ticket (#2254) the problem is actually with the config not being upgraded correctly, but we can't retroactively fix that so this fix is working around that problem.

Fixes #2254
closed 2023-06-18 by robustini