ExpressLRS/ExpressLRS #2263: Frames of radio channels frequently missing

m-rik-prog, 2023-06-20:
It is completely useless to make a protocol with high bps and Frame Rate values to lower the latency if then the channel packets are not transmitted with constant frequency and continuity but between one frame and the next there are often "gaps" (misses) of one, two, three, four whole frames.
The latency may be declarably low but the update of commands sent from the pilot to the model slows down and is not constant, continuous and smooth.

In order to verify that the missing packets are not due to the fact that in the meantime the value of the channels has not changed and therefore power, power consumption and radio frequency output can be reduced (as, for example, is the case with Jeti's EX Bus protocol that does not send new radio channel packets if the data is the same as the previous packet) fast automatic ramps have been created in the transmitter that continuously and quickly change the value of three radio channels.

Attached are the missing packets with ELRS protocol and for comparison the continuity and consistency of packets received with TBS protocol.

- TX hardware: RadioMaster TX16S
- External ELRS Module: RadioMaster Ranger Micro 2.4GHZ ELRS Module
- RX hardware: RadioMaster RP1 & RP2
- EdgeTX version: 2.8.4
- RxpressLRS firmware version: 3.2.1
- Packet rate 333Hz Full
- Telem Ratio: 1:32 or 1:16

[image]

[image]

pkendall64, 2023-06-20:
With ELRS we do not send in the telemetry window from the transmitter, so with your settings 1 in 16 packets will be missing from the transmit side. Also, you have not said if you have the LBT firmware installed, if you do then we listen on the channel before sending and if we detect that the channel is not clear then we do not send. If you set the power output to 10mW then we do not perform LBT and will send on every position.

m-rik-prog, 2023-06-20:
Yes, we use LBT, and the telemetry transmission power from the receivers has always already been set to the minimum: 10W
In contrast, the transmitter-mounted module transmission power has always been set to 25W and that is the minimum value that can be set, you cannot reduce to 10W.
I point out that in the oscilloscope images I sent you, not only one frame is missing every 16 but often two / three / four frames are missing consecutively.

I refer you to the image (this time enlarged) where it is visible in the main window of screen 46 that after the first frame the second frame is missing then the third frame is transmitted and immediately after that the fourth and fifth frames are missing and then the sixth, seventh and eighth consecutive frames are present. So there is not missing only one frame every sixteen but in this case there is missing three frames every eight.

[image]

There are then above the thumbnails of the previous screens where it can be seen in the thumbnail of screen 32 that as many as three frames are missing consecutively and the same is the case in the thumbnail of screen 39.
In screens 28 and 34 two frames are consecutively missing, etc. etc.

pkendall64, 2023-06-20:
It does not matter about the RX power. If the TX power is 25mW then the TX is not sending if it detects something on the channel it's about to use. We send 4 packets on each channel before hopping so if the channel is in use then you may miss up to 4 packets in a row.

m-rik-prog, 2023-06-20:
So the inability to bring the power below 25W is a bug in the lua app?

pkendall64, 2023-06-20:
> So the inability to bring the power below 25W is a bug in the lua app?

No, it's a hardware limitation. So devices can go down to 10mW some cannot. Usually modules with higher top-end output power (like the Ranger) cannot go down to 10mW.

m-rik-prog, 2023-06-20:
So by setting the frame rate of frames type 0x16 (RC channels packed) to 150Hz the frames could be transmitted at 150 / (1+4) = 30Hz? One every 33milliseconds?
If so, the ELRS protocol cannot be used on RC Heli or aerobatic aircraft. Command updates are too slow.
Thank you for this very important information!

pkendall64, 2023-06-20:
> So by setting the frame rate of frames type 0x16 (RC channels packed) to 150Hz the frames could be transmitted at 150 / (1+4) = 30Hz? One every 33milliseconds? If so, the ELRS protocol cannot be [...]

No, if you set it to 150Hz you get 150 packet slots per second. If you have 1:16 telemetry then every 16th packet is not sent from the TX, the TX listens for telemetry instead. In LBT mode, for each packet being sent from the TX it first listens to see if the channel is being used if it is then it does not send that packet. Channel hopping is done every 4 packets, this is just so we are not staying on the same channel.

JyeSmith, 2023-06-20:
Sounds like LBT is causing you some problems.  For LOS, a module that will go down to 10mW is fine, and will remove any regulatory issues.

You might also like to try the D250 mode.  This will give you a packet rate of 250Hz, but OTA it is 1000Hz.  So you get 4 repeat packets over 3 different frequencies.  This will give the RC data the best chance of being received with LBT enabled.

Sounds a bit extreme to say a 500Hz rate runs at 100Hz (in the worst case environement) but TBS CRSF will still hold 150Hz.

closed 2023-06-20 by pkendall64

m-rik-prog, 2023-06-21:
Thank you all, you have confirmed to me that the oscillographic graphs I posted to you in which it is clearly seen that the crossfire frames type 0x16 (RC channels packed) read on the serial output of the receivers it is normal that they can miss once (halving the frame rate), twice (bringing the frame rate to one third), three times (bringing the frame rate to one fourth) up to four times (bringing the frame rate to one fifth) and that this is not a defect but is expected and normal in the ELRS protocol. 

With ELRS the update of commands sent from the pilot to the model can slow down to five times the expected update rate.

It was important to know this

You also informed me that the lua app should allow you to reduce the transmit power below 25W down to 10W but unfortunately the current version of the lua app does not allow this.

JyeSmith, 2023-06-21:
> Thank you all, you have confirmed to me that the oscillographic graphs I posted to you in which it is clearly seen that the crossfire frames type 0x16 (RC channels packed) read on the serial output [...]

ELRS is used a lot in drone racing.  A community with an obsession for solid links, receiving every packet, and flying at events with maybe dozens of hands active causing interference.  These pilot also heavily log flight performance and link characteristics.  If what you are stating above was even remotely true, ELRS would not be used... but it is.

I can also guarantee you that if you are not seeing dropped packets with other links, then they are sending repeat packets from the rx.  Something ELRS doesn't do as flight controller such as Betaflight etc only expect real data.

I also gave 2 suggestions to try and provide what you would like to see (1:2 tlm and/or D mode).  How did they work for you?

> You also informed me that the lua app should allow you to reduce the transmit power below 25W down to 10W but unfortunately the current version of the lua app does not allow this.

As mentioned above, this is a limitation of the hardware you purchased.  There are plenty of tx modules capable of 10mW.

But in the end, if you do not want to provide ELRS support to your paying customers thats your call.  We can help, but your comments here and on other forums make it sound like you've already made up your mind.

SunjunKim, 2023-06-21:
> Yes, we use LBT
It's exactly how it works in LBT, for dropping packets with the existence of external interference (like WiFi, Bluetooth, etc), as mentioned before. Comparing an LBT policy vs. a blindly transmitting policy doesn't make much sense. They're not in the same ballpark. Try ISM2400 mode (=non-LBT), or if you have to use LBT try D-modes instead. 

> With ELRS the update of commands sent from the pilot to the model can slow down to five times the expected update rate.
This sounds like "I can win Lotto every time, because I have a non-zero possibility for hitting the six numbers" to me.
Stop exaggerating the extreme cases to be a general occasion, please.

m-rik-prog, 2023-06-21:
> But in the end, if you do not want to provide ELRS support to your paying customers thats your call. We can help, but your comments here and on other forums make it sound like you've already made up your mind.

Good morning,
If we were not interested in getting our FCUs to work properly even with the ELRS protocol, we certainly would not have purchased the transmitter module and receivers to do all the necessary tests and verifications to optimize the crossfire signal decoding of the radio channels and the encoding of the Crossfire telemetry transmissions by investing our own money but more importantly our own hours of work on the ELRS protocol.

If we did not care, we would not have even taken our time writing here to report what in our opinion and by comparison with all the other 25 radio protocols decoded by our FCUs seem to be problems and anomalies compared to all the other protocols.

Since we have to provide support for the one hundred thousand FCUs sold to users who should ask us for explanations, we need to understand and know whether the certain data we have seen and verified with our oscilloscopes (of which we have also sent you screenshots) are bugs related to the current firmware version that will then be fixed with future updates or whether instead they are data expected and established by the operation of the ELRS protocol.

However, this seems to be a problem for you, so we are discontinuing our reports and requests.
Bye

pkendall64, 2023-06-21:
I think you will find that it comes off a bit aggressive to start an issue tick with "It is completely useless to make a protocol with high bps..."
It would have been better to join the discord server and ask a few questions first. Then there could have been a quick back-and-forth to clear up any problems in your testing procedure.