ExpressLRS/ExpressLRS #2264: Crossfire protocol not bidirectional (non-standard)

m-rik-prog, 2023-06-20:
With the ELRS protocol, it is currently impossible to use "Integrations" (menus on the transmitter by which to read and change parameters from the flight controllers) because the implementation of the Crossfire protocol is not complete. Read and write requests sent by a LUA app on the transmitter do not reach the flight controllers.
LUA apps for now only work using TBS's Crossfire protocol.
This is a significant penalty of the ELRS protocol.

pkendall64, 2023-06-20:
If the Lua sends a CRSF packet with embedded MSP and the destination is set to BROADCAST or FLIGHT_CONTROLLER then the packet is send to the RX and the RX will forward the MSP to the FC.

closed 2023-06-20 by pkendall64

CapnBry, 2023-06-21:
If you would like assistance with whatever integration you're working on, then maybe you could provide details about it. The statement that "Protocol is not bidirectional" is false though. Look at the handset Betaflight Lua script which runs a bidrectional MSP connection tunneled through the RC link.

Due to telemetry bandwidth constraints, the connection can be lossy though, and there's only 1-2 packet buffer (depending on packet type) on the receiver side. It is very easy to overflow it if the connection is to be considered a lossless pipe with infinite bandwidth.

m-rik-prog, 2023-06-22:
Thank you for the availability,
From my LUA app I send toward the flight controller only two types of requests:
a) Reading value from a certain address.
b) Writing a new value to a certain address.

Through the LUA app I send a Crossfire frame (CRSF) composed as follows:

Sync Byte, Frame Length, Tipe, payload, CRC
0xC8, 0x0A, 0x28, 8 bytes of payload frame, CRC

With TBS module and receivers all this works smoothly and quite fast (the frame rate of TBS unlike ELRS is constant).

However, using the ELRS module and receivers purchased from RadioMaster the flight controller does not get any read or write requests.

Searching the net, I found on the public page visible to all the statement I have boxed in red:

[image]

I therefore stopped testing and verification believing that being public and visible by all of a very famous and widespread Open Source (like you) system was a verified, confirmed and certain statement.

However, now that you have told me that it should work I tried to see the source of your LUA app for configuring ELRS receivers and noticed that you use the same value of 0x28 as "Type".

So frames sent from my LUA app are probably intercepted by the receivers believing them to be read write frames from the internal configuration of the receivers but then discarded because they are in a format that does not conform to the expected format.

Maybe just change Type and instead of 0x28 as Type use another value not already used and defined by the Crossfire / CRSF protocol (EG: 0x22 to 0x27 or 0x29 to 0x3D).

Can you confirm me this theory?

CapnBry, 2023-06-22:
> Can you confirm me this theory?

I can confirm, your theory is not correct. The 0x28 type frame "device ping" is used by ELRS for the same purpose as TBS uses it for: detecting other connected devices. I do not know why Ardupilot has that information saying "[ExpressLRS] does not provide bi-directional telemetry" in their documentation. It could be from several years ago when we did not support full telemetry? In any case that statement is wrong or needs to define what bidirectional telemetry means. Mavlink? MSP over CRSF? IP over avian carrier?

The 0x28 type packet is forwarded to the flight controller, and any response from the FC is sent back to the handset. Our Lua script uses this to detect the receiver, and you'll even see Ardupilot and Betaflight show up in our device list, confirming that 0x28 is sent to the flight controller, and the response gets back to the handset.
[image]

Also, to be pedantic, the frame rate of ExpressLRS is also constant, however channels packets and telemetry packets each consume a frame's time slot. With TBS, the link rate is the channel rate and the telemetry comes "for free", but in our system telemetry can't go at the same time so this may appear as an inconsistent frame rate. Setting our TLM ratio to 1:2 effectively halves our channel update rate, but then there's just as many time slots for telemetry, making it somewhat equal to TBS's cadence. However, there doesn't tend to be enough telemetry to send to waste these time slots on it, so ExpressLRS dedicates more of them to sending channels packets. Here's an example of our connection running in 1:8 TLM ratio
[chan] [chan] [chan] [chan] [chan] [chan] [chan] [telem]
[chan] [chan] [chan] [chan] [chan] [chan] [chan] [telem]
[chan] [chan] [chan] [chan] [chan] [chan] [chan] [telem]

This may appear as an "not a constant rate" but instead is very constant. We also automatically switch to 1:2 TLM for a short time when the TX module has bidirectional data (bidirectional telemetry?) to send, which to someone only watching for channels data may appear not-consistent. This is why bidirectional data should not be sent during flight as it reduces the control update rate, but never below 1/2 the selected packet rate.

m-rik-prog, 2023-06-22:
Thanks for the information

Just for info: Searching with Google for "elrs telemetry protocol" the Ardupilot page appears as the sixth result.

My LUA app, when running, keeps sending packets like 0x28 for reading values until it gets a response.
Leaving the LUA app running and connecting the oscilloscope on the serial output of the receiver, no packets with Type equal to 0x28 ever appear, only the "RC channels packed" packets (Type = 0x16) and occasionally the Link statistics packets (Type = 0x14).
So at this point I can only speculate that there is either a problem in the firmware for RadioMaster's "Ranger Micro 2.4GHZ ELRS Module" transmitter module or more likely for RadioMaster's "RP1" & "RP2" receivers that do not let "pass" Type 0x28 packets or both.
Although I have already mentioned it, I remember that with TBS Modules and receivers, through the oscilloscope instead the type 0x28 packets sent through the LUA app are clearly visible and are recognized by the responding flight controllers. By putting in the flight controllers a break point at the time of acknowledgment, the controllers do not respond and by oscilloscope the same request packet keeps coming in on the serial output of the receivers.
So there is nothing left for me to do but to wait for the next firmware version with the hope that the problem will be solved.

As for the missing packets issue, I think the issue has been cleared up:
With the LBT versions (probably to meet all the stringent European regulations on radio emission limitations) by setting as "Packet Rate (Frame rate) the values 50Hz, 100Hz, 150Hz, 250Hz, 333Hz-Full, 500Hz or F500 the frame rate at which the radio channel packets come out of the receivers is not constant but up to 4 channel packets can also be missing for every 5 packets provided.
Hardly ever only one radio channel packet is missing but 2, 3 or 4 are CONSECUTIVELY missing.

This is not an "opinion" but an instrumentally verified certainty in the oscillographic graphs I have already sent you several times obtained by connecting the oscilloscope to the serial output of the receivers.
Only by selecting as "Packet Rate" (Frame rate) D250 or D500 then the radio channel packets are constant and always present every 4 milliseconds or every 2 milliseconds and the telemetry sent by the flight controller is regularly received and displayed on the transmitter.

However, with D250 and D500 the maximum number of transmitted channels is limited to only 12 channels of which the last 4 are low resolution.

As proof that the missing channel packets are intentional to limit radio frequency emission and to fall within the European emission maximum limits, I was written that by reducing the maximum emission power to 10mW, the radio frequency emission drops and this problem disappears and the packets are all sent.

However, even in this case, the firmware for the Radiomaster "Ranger Micro 2.4GHZ ELRS Module" prevents reducing the power to less than 25mW, so I cannot test what I have been told. I was told that other non-Radiomaster module models allow the power to be reduced to 10mW. I did not understand whether this is a hardware limitation of the Radiomaster modules or another possible firmware problem for the RadioMaster modules.

Again, I just have to wait for the next firmware version with the hope that it will be possible to reduce the transmit power to 10mW.

PS: Based on what you have told me, in the rest of the world where there are no limitations for radio frequency signal pollution these limitations should not be there.

In order to be able to transmit all 16 radio channels and at full resolution to be able to use the full functionality of the flight controllers, you have to select the Packet Rate "333Hz Full" but by setting 16 channels the radio channels are transmitted alternately in each frame so the update of each channel occurs at 166.5Hz every 6 milliseconds halving the actual frame rate. Add to this the fact that for LBT versions 4 packets of the radio channels can be missing every 5 lends and thus the frame rate can go as low as 33.4Hz every 30 milliseconds.
 
I think for you, verifying that what I have told you is true is very simple. Just connect a memory oscilloscope with a serial decoder (which I think like me all of you who develop and test your own firmware have) to the output of the RadioMaster receivers.

CapnBry, 2023-06-22:
I'm not sure how we're not sending packet 0x28 out to the flight controller in your testing, because if I see "betaflight" in my device list, that would not come from the receiver or transmitter module, that tells me that GUARANTEED we are sending packet type 0x28 out in our serial stream. There's some other key piece of information that's missing here. I would consider putting it on a scope myself but I don't think I need to, where else would "Betaflight" come from in our device list if we didn't forward it?

The rest of your stuff isn't relevant to this topic, but all the "missing packets" sound like it is a requirement of your local laws. If the RF channel is not clear then we do not transmit at all on that frequency until the next time we're on it. There's nothing we can do about this in software apart from ignoring the regulations and transmitting anyway. If that's what you'd prefer, just flash your ExpressLRS with the FCC regulatory domain. Problem solved.

Ignoring that, what do you think TBS does in their 12ch mode? They halve the packet rate. In 12ch mode, CH1-4 are sent at the displayed rate, and CH5-12 are transmitted at half rate, 4 channels per packet. This is the same as our system. Our system also has a 16ch mode where all the channels are sent at half rate, which they do not, so it isn't relevant to compare the two at that number of channels-- we've already won. Even with that, our 333Hz (166.6Hz) mode is still faster than their fastest mode, but carrying twice the number of channels.

I do not care to discuss your packet rate math in this thread though, as that is not relevant to the discussion at hand. I do believe you're wrong about us not sending packet 0x28 to the flight controller though because there's only one explanation for where "betaflight" or "ardupilot" shows up in my connected device list. I would suggest making sure your telemetry is working (not just link statistics, but actual telemetry like position and VBAT are showing up) then see if it works. I don't remember there being anything blocking our code from forwarding the packet when the link to the FC is down but I may be forgetting something. Verifying the telemetry link is established is important because our system needs to "sync up" with the RX, and it may take up 10-20 seconds for this to occur if the TX was in the middle of telemetry reception with a previous RX. Much longer at low packet rates and low telemetry ratios.

CapnBry, 2023-06-22:
And just for completeness, here's the CRSF packet we send to the TX module from our Lua to send the 0x28 command `EE 04 28 00 EA 54`

Also here's a debug tool I used originally when reverse engineering the protocol, it only expects there to be 2x devices so it might take a couple close/open of the script but I see the betaflight device ping response here most of the time (the other times it is just off the screen)
```lua
-- TNS|CRSF Dump|TNE
-- Hacked together by CapnBry for the ExpressLRS project from crossfire.lua

local devices
local lineIndex
local deviceId
local fieldId
local fieldChunk
local fieldTimeout

local function createDevice(id, name)
  local device = {
    id = id,
    name = name,
    timeout = 0
  }
  return device
end

local function getDevice(name)
  for i=1, #devices do
    if devices[i].name == name then
      return devices[i]
    end
  end
  return nil
end

local function selectDevice(step)
  lineIndex = 1 + ((lineIndex + step - 1 + #devices) % #devices)
end

local function fieldGetString(data, offset)
  local result = ""
  while data[offset] ~= 0 do
    result = result .. string.char(data[offset])
    offset = offset + 1
  end
  offset = offset + 1
  return result, offset
end

local function parseDeviceInfoMessage(data)
  local id = data[2]
  local name, i = fieldGetString(data, 3)
  local device = getDevice(name)
  if device == nil then
    device = createDevice(id, name)
    local serialNum = string.format("%02x%02x%02x%02x", data[i], data[i+1], data[i+2], data[i+3])
    i = i + 4
    local hardwareId = string.format("%02x%02x%02x%02x", data[i], data[i+1], data[i+2], data[i+3])
    i = i + 4
    local softwareId = string.format("%02x%02x%02x%02x", data[i], data[i+1], data[i+2], data[i+3])
    i = i + 4
    device.sn = serialNum
    device.hwid = hardwareId
    device.swid = softwareId
    device.fldcnt = data[i]
    i = i + 1
    device.ver = (#data >= i) and data[i] or 0
    devices[#devices + 1] = device
  end
  local time = getTime()
  device.timeout = time + 3000 -- 30s
  if lineIndex == 0 then
    lineIndex = 1
  end
end

local function parseParameterInfoMessage(data)
  local s = ""
  -- Start of packet is always { EA, [deviceid], [fieldid] }
  for i = 3, #data do
    s = s .. string.format("%02x", data[i])
  end
  if fieldId == 3 then
    devices[lineIndex].msg0 = s
    devices[lineIndex].msg1 = string.sub(s, 24)
    fieldTimeout = 0
  elseif fieldId == 1 then
    devices[lineIndex].msg1 = s
    -- request fieldId X second
    fieldId = 3
    fieldTimeout = 1 
  end
end

local devicesRefreshTimeout = 0
local function refreshNext()
  local command, data = crossfireTelemetryPop()
  if command == nil then
    local time = getTime()
    if time > devicesRefreshTimeout then
      devicesRefreshTimeout = time + 500 -- 5s
      crossfireTelemetryPush(0x28, { 0x00, 0xEA })
    elseif fieldTimeout ~= 0 and time > fieldTimeout then
      crossfireTelemetryPush(0x2C, { deviceId, 0xEA, fieldId, fieldChunk })
      fieldTimeout = time + 200 -- 2s
    end
  elseif command == 0x29 then
    parseDeviceInfoMessage(data)
  elseif command == 0x2B then
    parseParameterInfoMessage(data)
    --fieldTimeout = 0
  end
end

local function setColorVar()
  if LCD_W == 480 then
    barColor = (TITLE_BGCOLOR or TITLE_BGCOLOR)
    titleAdditionAttr = (WHITE or MENU_TITLE_COLOR)
    progressBarColor = (GREEN or WARNING_COLOR)
  else
    barColor = GREY_DEFAULT
    titleAdditionAttr = INVERS
    progressBarColor = 0
  end
end

local function init()
  lineIndex = 0
  fieldTimeout = 0
  devices = {}
end

local yln = 0
local function println(x, label, value)
  if x == nil then
    yln = 0
    return
  end
  lcd.drawText(x, (yln+1)*14, label .. ': ' .. tostring(value), 0)
  yln = yln + 1
end

local function run(event)
  if event == nil then
    error("Cannot be run as a model script!")
    return 2
  elseif event == EVT_VIRTUAL_EXIT then
    return 2
  elseif event == EVT_VIRTUAL_NEXT then
    selectDevice(1)
  elseif event == EVT_VIRTUAL_PREV then
    selectDevice(-1)
  end

  lcd.clear()
  --lcd.drawScreenTitle("CRSF Config Proto", 0, 0)
  if #devices == 0 then
    lcd.drawText(4, 28, "Polling for devices...")
  else
    for i=#devices, 1, -1 do
      local d = devices[i]
      local attr = (lineIndex == i and INVERS or 0)
      if event == EVT_VIRTUAL_ENTER and attr == INVERS then
          --crossfireTelemetryPush(0x28, { d.id, 0xEA })

          deviceId = d.id
          fieldChunk = 0
          -- request fieldid 1 first
          fieldId = 1
          fieldTimeout = 1

          crossfireTelemetryPush(0x2C, { deviceId, 0xEA, fieldId, fieldChunk })
          -- return "device.lua"
      end
      local y = 4 * (i - 1)
      lcd.drawText(0, (y+1)*8, string.format("%s (%02x)", d.name, d.id), attr)
      lcd.drawText(0, (y+2)*8, string.format("Sn:%s Hw:%s",
        d.sn, d.hwid), 0)
      lcd.drawText(0, (y+3)*8, string.format("Sw:%s Cnt:%d V:%d",
        d.swid, d.fldcnt, d.ver), 0)
      lcd.drawText(0, (y+4)*8, "0:" .. (d.msg0 or ""), 0)
      -- lcd.drawText(0, (y+5)*8, "1:" .. (d.msg1 or ""), 0)
    end
  end
  refreshNext()
  return 0
end

return { init=init, run=run }
```

m-rik-prog, 2023-06-23:
Thanks for all the considerations and evaluations and thanks for the LUA source.

I tried the LUA source that displays the transmitter and receiver module information (even though the hardware versions appear to be identical) but no packets come out on the receiver serial output.

Since you are certain, however, that it can work because it works with Betaflight, and because you mentioned "source" and "destination" addresses, I tried using instead of the "Braodcast" free Type values (0x22 to 0x3D) that work with TBS, the "Extended header frames" Type values (I used 0x40) and was finally able to get the read request and write request packets to appear on the receivers' serial outputs.
Now all sent packets and all received and transmitted frames are longer than two bytes (+origin and +destination) and I had to increase in the firmware of my flight control units the size of the vectors in Ram by also changing the read positions of the values and these two extra bytes will make the CPU work a little bit harder, but the important thing is that a solution was found (which by the way also works with TBS modules and receivers).

Since you are sure that Type 0x28 "Braodcast" also comes out on the serial bus of the other ELRS receivers as it comes out of the TBS receivers it is obvious that there is a problem in the specific firmware version made for Radiomaster's P1, P2, P3 receivers, but still a workaround / escamotage is possible and works.
Thank you!

Now as "problem" remains only that of frame Type 0x14 "Link statistics":

Payload:
typedef struct {
uint8_t up_rssi_ant1; // Uplink RSSI Ant. 1 ( dBm * -1 )
uint8_t up_rssi_ant2; // Uplink RSSI Ant. 2 ( dBm * -1 )
uint8_t up_link_quality; // Uplink Package success rate / Link quality ( % )
int8_t up_snr; // Uplink SNR ( dB )
uint8_t active_antenna; // number of antenna
uint8_t rf_profile; // enum 4fps = 0 , 50fps, 150hz
uint8_t up_rf_power; // enum 0mW = 0, 10mW, 25 mW, 100 mW, 500 mW, 1000
P.8
mW, 2000mW, 250mW
uint8_t down_rssi; // Downlink RSSI ( dBm * -1 )
uint8_t down_link_quality; // Downlink Package success rate / Link quality ( % )
int8_t down_snr; // Downlink SNR ( dB )
} libCrsf_link_statistics_s

For "rf profile" TBS has been using value 1 for 50hz and value 2 for 150hz for years.
ELRS uses value 2 for 50Hz (value 3 is free and unused).

When I pointed this out that prevents Crossfire/CRSF decoding from automatically recognizing the frame rate, Paul Kendall replied, "We are not TBS or Crossfire. The values we use match the rates that we have in ELRS." (I find the answer strange because as the serial output protocol used by ELRS receivers is used the Crossfire serial protocol).

Because of this reason my controllers do not work properly with ELRS at 50Hz and the values calculated by timers for "Fades" which can increase the value of Frame Losses which in turn raise the value of "Holds" (total lack of signal for more than 45 consecutive frames) all skyrocket.

Therefore, if you think it is right that the rf_profile value of ELRS for 50Hz is identical to the value used by TBS for 150Hz (despite the fact that the value 3 is free and unused) I need to know how else my flight controllers can automatically recognize whether there is a TBS receiver or an ELRS receiver connected.

Thanks!

pkendall64, 2023-06-23:
What I mean by my comment is that TBS specifies the protocol and syntactic meaning of the structure but not the semantic meaning of the values in the fields of the structure.

You can detect if you are talking to an ELRS device when send a device ping packet we will respond with device information that contains the bytes 0x454C5253 in the serial number field which is the characters 'E', 'L', 'R' and 'S'.

Hope this helps.

m-rik-prog, 2023-06-23:
Thank you for the feedback of the clarification and the solution.

Going a little more specific though:

Since the flight controllers are connected to the receivers only by the serial cables over which the Crossfire protocol travels, what is the Type of the data packet to be sent to the receiver to receive in reply and what should be the content of the Request Payload?

A simple note: For automatic recognition of various protocols, the first byte of the radio channel frame is normally used. For example, Spektrum's SRLX receivers use 0xA2, or 0xA5 as the first Byte, SRLX Multiplex receivers use 0xA1, etc. etc. Once the baud rate of the serial signal is recognized, it is then very easy at the same baud rate to automatically distinguish any protocol from any brand of radio systems. However, it seems that with Crossfire it has been established that the first Byte (Sync Byte: 0xC8) is always the same even for different systems. In my opinion, this is a flaw in the Crossfire protocol.

CapnBry, 2023-06-23:
I would say the flaw in the Crossfire protocol is that the first byte is supposed to change based on the destination. The first byte can be anything! It makes it very hard to validate frames if they can start with any byte. 😄

Regarding our RF mode IDs, we have our own list and TBS has their own list. That's not going to change because TBS isn't interested in maintaining a comprehensive list of all 3rd party RF modes as part of their protocol.

You really need to stop immediately going to the conclusion "I couldn't do this, so therefore the firmware on my devices is flawed. I await the fix.". It comes of as being a total asshole. I'm trying to assist you with your technical problems but I do not appreciate the tone.

> what is the Type of the data packet to be sent to the receiver to receive in reply and what should be the content of the Request Payload?

The type of which data packet for what? For the device info? That's the data that comes back from 0x28, `CRSF_FRAMETYPE_DEVICE_INFO = 0x29`

brandonrc, 2023-06-24:
@ m-rik-prog,

Thank you for your contributions to our community. We understand and respect the responsibilities and challenges of running a business. However, let me underline the importance of respectful and constructive communication in our interactions.

Our community is expansive, with 16,823 members on Facebook and 18,560 members on Discord, and it's constantly growing. Each one of us is volunteering our time, skills, and passion to help others and enhance the software. We aren't doing this for financial gain, but out of love for this community and the desire to create something we can take pride in. This does not, in any way, diminish the value of our time or the quality of our work.

It's not about seeking 'caresses, kindnesses, and niceties', but about fostering an environment of mutual respect and understanding. It's not about us being 'touchy', it's about creating a community where people feel valued and where their contributions are acknowledged.

The tone of your messages has been less than appreciative. It's important for you to understand that if you feel your time is more valuable, you are always free to create your own RC link for your product. But when engaging with this community, we ask for your understanding and respect.

We look forward to more productive and respectful interactions in our future discussions. Thank you again for your involvement and contributions.

Best,

brandonrc, 2023-06-24:
P.S. To emphasize, @CapnBry  and @pkendall64, who you've been less than respectful to, have been going the extra mile to help you out. I'm certain their time is just as valuable, if not more so than yours, given their significant contributions to this community.

CapnBry, 2023-06-24:
And just as the final bit of testing, I hooked a receiver up and dumped the serial data coming out of it when I did the 0x28 type packet from Lua on the handset, and 😱 . A type 0x28 packet comes out of the receiver, which would be going out to the flight controller (channels packets are omitted for clarity).
```
RSSI= -52dBm LQ=100 mode=9
RSSI= -50dBm LQ=100 mode=9
RSSI= -50dBm LQ=100 mode=9
RSSI= -55dBm LQ=100 mode=9
RSSI= -57dBm LQ=100 mode=9
RSSI= -53dBm LQ=100 mode=9
RSSI= -51dBm LQ=100 mode=9
RSSI= -52dBm LQ=100 mode=9
RSSI= -51dBm LQ=100 mode=9
Unknown: 0xc8 0x4 0x28 0x0 0xea 0x54 <-- packet 0x28 to FC
RSSI= -51dBm LQ=100 mode=9
RSSI= -54dBm LQ=100 mode=9
RSSI= -52dBm LQ=100 mode=9
RSSI= -50dBm LQ=100 mode=9
RSSI= -54dBm LQ=100 mode=9
RSSI= -51dBm LQ=100 mode=9
RSSI= -50dBm LQ=100 mode=9
RSSI= -50dBm LQ=100 mode=9
```

I entered this conversation because when ExpressLRS isn't doing something right, then I work my hardest to fix it. Especially in the case of working with developers of other projects, I feel like at that professional level it is easy to resolve any issue due to the depth of engineering knowledge on both sides working to determine where the problem lies. 

Every piece of evidence I have found leads me to believe packet 0x28 is working as per our mutual expectations. The format of the request coming from Lua (`EE 04 28 00 EA 54`) is what I believe it should be. The packet going to the FC (`C8 04 28 00 EA 54`) is present, and if the FC responds with a type 0x29 packet, it shows up on the handset. There's nothing else I can do to assist you. Maybe get some packet dumps going between the handset and TX module to verify the outgoing request looks right? I dunno, but you've made it clear my time isn't worth anything to you so I'll leave you to it.

m-rik-prog, 2023-06-28:
Again, thank you for the further testing!

I had no doubt and was sure (I even wrote it down) that you could read the Type 0x28 broadcast frames on the output of your receivers.

However, what I am saying, and repeating, is that with my "Ranger 2.4GHz ELRS Module" LBT from RadioMaster and my P2 LBT receivers from RadioMaster both updated via "ExpressLRS Configurator" to the current 3.2.1 "EU_CE" version, the 0x28 broadcast packets are not coming out of the serial output of my RadioMaster receivers.

Have you tried using RadioMaster's "Ranger 2.4GHz ELRS Module" LBT updated via "ExpressLRS Configurator" to the current 3.2.1 EU_CE version and using a RadioMaster's P2 LBT receiver updated via "ExpressLRS Configurator" to the current 3.2.1 EU_CE version?

I just need to know this because if the tests were done this way, then my modules and receivers are defective and I will ask for their replacement before the warranty expires or I will buy new ones.

As an aside, however, I also said that I still solved it by using extended header frames Type 0x40.
So the problem is definitely solved for me anyway.

I also successfully used the 0x28 request in the opposite direction (from the flight controller to the receiver so as to distinguish whether the value 2 is related to TBS or ELRS and now the trend of the curves of the recorded values of frame losses and frame rate turn out to be correct (except when there are missing frames as expected for LBT / EU_CE versions).

I take advantage of another question: For decoding the CRSF serial signal, I have the UART set to 416.666bps as for TBS and I see that it always works fine and no UART decoding error is ever created. However, I read on your site that you talk about 400.000bps. The question is: Is it a coincidence that the UART never reports any kind of error by decoding a 400,000bps signal at 416.666bps or is it a typographical error and where it was written 400Kbps it should have been written 417Kbps instead?

On the other hand, regarding the phrase "you've made it clear my time isn't worth anything to you," I don't understand why in this group anything that is asked or said is always considered a personal offense. The reported sentence is not true, and I ask for the courtesy not to put things in my mouth that I never said (just reread what I wrote), because it is grossly incorrect. Such things are done in political discussion forums against opponents, I am not an opponent, I have never said such a thing and this is a technical and not a political forum where you only try to discredit your opponents.
I for my part am sure that I have never offended or insulted anyone, that I have always appreciated and thanked for the technical answers I have received, and that I have also justified why in my posts I try to be concise by going straight to the point of the issues, without a lot of frills and for this reason I may therefore seem to some people brusque and grumpy. I feel that I have always been fair even if for others this is evidently not the case. There is nothing I can do about that. I repeat, we are not in a love dating forum or even a general discussion forum, but in a technical forum.

CapnBry, 2023-06-28:
I just flashed a Ranger using Configurator to 3.2.1 EU_CE along with an RP2 receiver to the same version. I get the same result, which is that the 0x28 packet shows up on the receiver. 

I do not believe there is any problem with your hardware. No hardware problem would prevent just one single packet type from going through the system as designed. You claim to be knowledgeable with regards to technical details so you would know that this would not be a hardware issue.

There is some fundamental software difference between our testing methods. I get the 0x28 packet to be sent simply by launching the ExpressLRS Lua on the handset (elrsv3.lua) while connected at 250Hz Std TLM Ratio, and see the packet come out  an FTDI cable connected to the receiver. Maybe you'd like to share the code you're trying to run that's not working?

I would very highly not recommend building your system with a workaround for an inability to get the proper way working. There is no guarantee that ExpressLRS will exhibit the same behavior with your 0x40 type packet **especially** with CRSF protocol changes coming soon.

The baud rate for our receivers is 420000 by default. This matches the existing baud rate used by TBS in Betaflight and iNav. I believe KISS prefers 400000 for their receivers. The TX module's baud rate is 400000 (one of many the TX module supports) which was what TBS had existing in OpenTX.

CapnBry, 2023-06-28:
Wait a second, someone just pointed out you said `0xC8, 0x0A, 0x28, 8 bytes of payload frame, CRC` but what are your 8 bytes of payload? The protocol I quoted above, and the test Lua I provided, this sends 2 bytes of payload in the 0x28 packet. 

What **exactly** are you sending?

FHermisch, 2024-01-12:
Hi, some days ago I implemented a "Flight Controller" (in CRSF terms) on a microcontroller connected to an ELRS receiver.
I managed to transmit a menu for the LUA script, providing a basic on/off for an LED. The LED is now controllable from the remote controller by switching it on/off via the script.
So - works for me...
Regarding the discussion above: I think key was to first send a proper device-info response on a broadcast. Afterwards all parts of the chain "know" that there is a flight-controller who is interested in messages and forwards them.