ExpressLRS/ExpressLRS #2275: Bluetooth Joystick broken in MacOS Ventura

phantom-8, 2023-06-27:
## Current Behavior
BLE Joystick is not working under macOS Ventura.  The "ExpressLRS Joystick" is discovered when searching for bluetooth devices in Preferences.  When clicking the "Connect" button, it's indicated as connected.  After around 30 seconds, the joystick device will be disconnected automatically.  During these 30 seconds connected duration, the joystick is not recognised by any applications.  This problem only occurs under macOS Ventura.  It's working perfectly when I boot my mac to Monterey.

## Steps to Reproduce

1.  Boot to macOS Ventura

2.  Under Bluetooth Preference, search nearby Devices

3.  The "ExpressLRS Joystick" is discovered.  Click "Connect"

4.  The joystick device is connected, but cannot be recognised by any applications that supported gamepad/joystick.

5.  After around 30 seconds, the joystick device will be disconnect automatically.

6.  Try to click "Connect" again and whole cycle above repeats.

## Possible Solution (Not obligatory)

The BLE Joysticks works fine under macOS Monterey.  Problem only occurs in Ventura.

## Details

## Your Environment

- TX hardware:  DIY 2400 TX ESP32 SX1280 E28
- ExpressLRS version (TX & RX MUST MATCH):  3.3.0-RC1

pkendall64, 2023-06-27:
This is working perfectly for me on Ventura 13.4.1 using 3.3.0-RC1.

phantom-8, 2023-06-28:
Just upgraded to macOS 13.4.1.  Problem still persisted.  Not sure what's cause, but it worked when I rebooted to macOS Monterey.

haandol, 2023-12-04:
same issue here on Ventura 13.5.2.

Askhento, 2024-01-08:
doesn't work on BigSur 11.7.10

closed 2024-04-19 by github-actions[bot]