ExpressLRS/ExpressLRS #2284: Airport on Lilygo / TTGO 433 MHz Hardware / ELRS 3.3.0-RC2

Target0815, 2023-07-02:
TX and RX flashed with 3.3.0-RC2

100 Hz / TLM 1:2. In Web-GUI set to Airport / 4800 Baud.

Arduplane FC set SERIAL1 to 4800 Baud, Mavlink 2

TX via USB connect to Laptop

When trying to connect with 4800 baud in Mission Planner, the Airport TX reboots immediately and a connection is not established. This was already the case with 3.3.0-RC1.

What is wrong with the above settings? What settings are recommended and guaranteed to work?

pkendall64, 2023-07-02:
If you read this page https://www.expresslrs.org/software/airport/ then you will see that the maximum baud for 100Hz is 2400 baud.
It could be that the buffers are overflowing and causing the software to crash and reboot.

Target0815, 2023-07-02:
Is now at 2400 baud and still boots when trying to connect.

Target0815, 2023-07-02:
Next try with 2G4 hardware (HM ES24TX Slim and BetaFPV EP1), same settings as above, 2400 baud.

With this, the connect works right away. However, the connection is frighteningly slow, it takes several minutes to load the parameters.

Compared to mLRS (Lora, 868 MHz), which is set to only 19 Hz, but works with a baudrate of 57.600 and has a special Mavlink handling, these are worlds. Here the parameters are loaded in a few seconds.

Also in comparison with RFD868, which is about on mLRS level.

All comparisons with always the same FC and the same laptop or MP version.

Either Airport is still buggy and really that slow, or I am doing something thoroughly wrong. I would therefore ask that someone who REALLY knows about Airport, once reported here ...

Target0815, 2023-07-04:
No one seems to be more closely involved with Airport ... I would like someone to contact me, where all this is no problem and we then go through the configuration once together.

pkendall64, 2023-07-04:
The devs are extremely busy at the moment. But if you call I said you should ask on discord and others that have it setup and working will likely help out.

Target0815, 2023-07-05:
@pkendall64 

Hi Paul, I understand that the devs are busy, but there are certainly users who are concerned with Airport.

Anyway, this is about an issue, most likely it is a bug in ExpressLRS and for that the issue tracker on Github is the appropriate place and not a confusing chat entity like Discord ...

So I ask you to remove the closed label please.

pkendall64, 2024-04-06:
Can we close this issue now or is still a problem?
I see that you have been active in looking at Airport and mavlink-rc on discord.

closed 2024-04-06 by Target0815

xros, 2024-04-19:
> Is now at 2400 baud and still boots when trying to connect.
Did you make it now? The lowest baud rate on Mission Planner is 1200. Did you try it?