ExpressLRS/ExpressLRS #2285: DIY Tx power will not go beyond 25mw when built using FCC mode

asudhir1947, 2023-07-04:
## Current Behavior
I can select the TX power all the way upto 500mw on my DIY tx module, sitting inside a Frsky XLite, using the regular v3.0 Lua

I updated to RC2 today and the TX power would not go beyond 25mw
## Steps to Reproduce
Update the DIY Tx module to RC2
Run Lua
Initially the Lua says error on the power level, but I can then change it to 25mw or 10mw and set it there.

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

 Issue does not affect me as I normally use 25Mw, but I dont want to be stuck there
 No

## Your Environment

- TX hardware: DIY Elrs TX module with build suggestion from tweetfpv
- RX hardware: happymodel ep1
- Handset model: frsky xlite
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 
- Packet Rate: 500hz
- Telemetry Ratio: 1:128
- user_defines: None

pkendall64, 2023-07-04:
I would say that you have flashed the wrong target. The DIY Mini does only 10 and 25mW.
You may be looking for the DIY E28 target.
I can't tell though because you have not given exact information about the target i.e. the target name you flashed from configurator.

asudhir1947, 2023-07-05:
Hi, you are right, I used the DIY Mini, that is the only target that supports ESP32 SX1280 and can also confirm I see mini in the elrs lua script, would the E28 target be good enough for my target ? I built the tx using the tutorial from tweetfpv, so not sure if the E28 would be good enough

pkendall64, 2023-07-05:
If you previously had 250mW then you are probably after this target
[image]

asudhir1947, 2023-07-05:
Thank you that worked, able to change output power level to 250mw now once again

closed 2023-07-05 by asudhir1947