ExpressLRS/ExpressLRS #2286: HM ES24TX Pro Blank ELRS Lua Script when using 3.3.0-RC2

UkrainianTym, 2023-07-04:
## Current Behavior

When updating HM ES24TX Pro to 3.3.0-RC2, the Lua script inside Tools in TX16S is blank and is only showing EXIT.  TX model is recognized on top but everything else is blank. TX binds without an issue but can't adjust power level, enable wifi, etc. Using the same ELRSv3 Lua Script and downgrading to ELRS version 3.2.0 on HM shows all information as usual, but no longer binds since the receiver version is mismatched. 

## Steps to Reproduce

1. Update HM ES24TX Pro to 3.3.0-RC2
2. Open ELRS Lua Script on TX16s Mark II
3. Model displayed on top and EXIT on bottom. Blank otherwise.

## Details

Need to take the TX apart to downgrade to the working 3.2.0 version since Lua script inside Tools is blank and can't enable WIFI otherwise.

## Your Environment

- TX hardware: HM ES24TX Pro
- RX hardware: BetaFPV SuperD 2.4 GHz
- Handset model: RadioMaster TX16S Mark II
- Edgetx version: 2.8.4
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0-RC2 (binds)

pkendall64, 2023-07-05:
Only the major part of the version number needs to match. i.e. all version 3.x are compatible.
I think you have some other issue going on. Best place to as is discord as this is not a software defect.

closed 2023-07-05 by pkendall64

rdahlin, 2023-07-24:
I saw that this issue is closed but I just upgraded from 3.0.x to 3.3.0 on a TX16S MK1 and I have the described behavior ie a blank screen (after a blink "Loading...") with "HM ES24TX Pro" in the top left corner and "C" or "-" if connected / not connected to a RX. I also get the exit at the bottom when scrolling the wheel.

I have tried to download the current Lua script and put it on my radio and also remove the .luac file but I still have the same result.