ExpressLRS/ExpressLRS #2296: BetaFPV SuperP 14Ch Receiver wrong Voltage reading (RxBt)

3DRacer, 2023-07-08:
The Voltage reading at VBAT/6s Pin is mutch to high. The ADC also seems to be at the end at about 8.4V. Higher values don't change anything anymore. BetaFPV does not seem to mean 6s LiPo at all.

Device: BetaFPV SuperP 14Ch Receiver
elrs 3.3.0-RC2

pkendall64, 2023-07-08:
https://github.com/ExpressLRS/ExpressLRS/assets/512740/5f5cbff5-f22a-4113-8bb2-6894428b2d66

Heres a video showing the SuperP going from 25V to 3V input on the VBAT pins. Note that the receiver is not powered by the VBAT connector.

3DRacer, 2023-07-09:
Thanks for the video. I tested in same condition. My receivers have an HW issue. I will contact betafpv. I bought several of these directly from there shop. all have same behavior.

[image]
[image]
[image]

pkendall64, 2023-07-09:
Can you start the receiver in wifi mode and go the `/hardware.html` page and verify that the options for VBAT are as expected.
[image]

3DRacer, 2023-07-09:
[image]

3DRacer, 2023-07-09:
I suspect wrong resistors in the ADC divider. Can someone please measure the resistance between VBAT and GND on their SuperP? Mine are all 77.8 KΩ.

here again all values of the HW side

[image]

pkendall64, 2023-07-10:
I spoke with BETAFPV today and the values we have used here are based on the samples we were sent, and in the production unit the resistor values have changed.
You could just change the `285` in the scale to `64` and I think that will be closer. We will have to get some new devices so we can properly calibrate them.

3DRacer, 2023-07-10:
I set the VBat scale value to 64. This produces an output of 162 Volts at 8.00V on the VBAT. I then played with the values on VBat scale and the value 1280 fits best.
Thus, VBAT and telemetry between 0.60V and 8.00V fits well.
BUT: Voltages above approx. 8 volts are not recorded by the ADC. The original problem still exists.

3DRacer, 2023-07-10:
thanks man. you made my day. short test before sleeping time shows it work. Changes of values up to 25 Volts at VBAT Pin. Tomorrow i will find out optimal values for span. Thanks a lot

pkendall64, 2023-07-10:
> thanks man. you made my day. short test before sleeping time shows it work. Changes of values up to 25 Volts at VBAT Pin. Tomorrow i will find out optimal values for span. Thanks a lot

You're welcome. BETAFPV are sending production units to the devs so we can calculate a proper set of calibration values. So I'll update this ticket once I get it.

3DRacer, 2023-07-11:
Here are my determined values for the SuperP.
Measurement results (Calibrated DVM):
VBAT ----- RxBt
 5.00V ---- 5.0
10.00V --- 10.0
15.00V --- 15.0
20.00V --- 20.0
24.00V --- 23.8
VBAT > 24 volts results in non-linear values.

[image]

pkendall64, 2023-07-18:
Hi, I've received some production units from BetaFPV and performed the calibration and come up with some new values.
Added in PR #2307

closed 2023-07-18 by 3DRacer