ExpressLRS/ExpressLRS #2297: AirPort serial connection not working

sebaszz, 2023-07-12:
Current Behavior
AirPort serial connection not working

Steps to Reproduce
Both (TX & RX) updated to 3.3.0-RC1 

Paired using binding phrase

Activated AirPort functionality via WiFi on TX and RX

Set baudrate to 4800 for TX and RX, packet rate to F100Hz

Serial RX connected to FC (serial3, baudrate 4800, options 0, mavlink 2)

Connected to TX via usb on PC to mission planner selected active com port baudrate 4800

Details
I want to establish a serial connection between TX en PC it's rebooting when trying to connect
TX indicates connected tot RX

I did not make changes to the code.

Your Environment
TX hardware: BETAFPV TX micro 868Hz
RX hardware: BETAFPV RX 868Hz
ExpressLRS version (TX & RX MUST MATCH): 3.3.0-RC1
Packet Rate: F100
Telemetry Ratio: 1:2
user_defines: default

Target0815, 2023-07-13:
Sounds familiar to me ... https://github.com/ExpressLRS/ExpressLRS/issues/2284

By the way, on Discord the airport section has been removed if someone comes to it with a hint.

JyeSmith, 2023-07-14:
Try updating to RC2.  There was a potential fix for this https://github.com/ExpressLRS/ExpressLRS/pull/2257

sebaszz, 2023-07-14:
thanx! 

it's working now.

upgraded to rc2. started to receive some data but still time out.
discovered there is also packet rate "100hz full" besides 100hz. 
Saw this in the documentation but didn't noticed it while setting up

closed 2023-07-14 by sebaszz