ExpressLRS/ExpressLRS #2314: 3.3.0 final release shows unclear version in LUA script

jiri-jirus, 2023-07-22:
Hello, after final release 3.3.0, lua script reports "3.x.x-maintenance EU868" for in lua script on my TX16s. (also on my ES24TX) This way, I dont know which version is loaded on module. I suppose this is not desired behaviour as I expect to see "3.3.0"

[image]

pkendall64, 2023-07-22:
We are aware of this, there was a bit of an issue where the 3.x.x-maintenance branch was re-pushed to the repository after the 3.3.0 release with the same git hash so it overwrote the artifacts. They are 100% the same though.

closed 2023-07-22 by pkendall64

offer-shmuely, 2023-07-24:
maybe we can get 3.0.1 with the fix?
we do like to remember, in the future, what the version that we installed

jiri-jirus, 2023-07-24:
> maybe we can get 3.0.1 with the fix? we do like to remember, in the future, what the version that we installed

probably 3.3.1