ExpressLRS/ExpressLRS #2315: 3.3.0 100Hz Full only 14 channels

jasc76, 2023-07-22:
## Current Behavior
I only got 14 channels. 

## Steps to Reproduce
Hyppymodel 2.4 EP Dual
flash from 3.2.1 to 3.3.0 channels dont work
flashed back to 3.2.1 channels work again

## Possible Solution (Not obligatory)

## Details

100Hz Full in 16/2 mode

## Your Environment
- TX hardware: Jumper AION 2.4
- RX hardware: Happymodel EP Dual
- Handset model: X20s
- OpenTX version (including nightly number): Ethos 1.4.8
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0
- Packet Rate: 100hz
- Telemetry Ratio: 1:4
- user_defines:

pkendall64, 2023-07-22:
If this is with CRSF output then that is correct as the last 2 channels of CRSF are RSSI dBm and LQ.
I'm not sure what you are actually reporting as the bug here! Is that you only get 14 channel output in 3.3.0 or no channels at all?

jasc76, 2023-07-23:
I only get 14
I didn’t change anything, I just updated to 3.3
Yes output is crsf. But thats the same on 3.2 So how do I have to set it on 3.3 ?

pkendall64, 2023-07-27:
> I only get 14 I didn’t change anything, I just updated to 3.3 Yes output is crsf. But thats the same on 3.2 So how do I have to set it on 3.3 ?

CRSF output has always only ever had 14 channels from the transmitter and the channel 15 and 16 are RSSI dBm and LQ generated on the receiver side.

closed 2023-07-27 by pkendall64