ExpressLRS/ExpressLRS #2329: Model Match greater than 14 hangs elrsV3.lua

analogsoftware, 2023-07-26:
## Current Behavior

Assigning a model match number greater than 14 hangs elrsV3.lua script, preventing further configuration of ELRS parameters. I do not know if it is the actual number "15" or the fact that I have 15 actual receivers configured - I presume it is the latter and the actual number is irrelevant as long as it is between 0-63.

## Steps to Reproduce

1. Create 14 ELRS models and verify you can access the elrsV3.lua script to configure parameters.

2. Create one more model, and elrsV3.lua hangs on open.

## Possible Solution (Not obligatory)

Disabling model match on the 15th model allows the elrsV3.lua script to open.

## Details

I have a total of 80 models on this transmitter, using various other protocols mostly DSMX and FrSky.

## Your Environment

- TX hardware: Radiomaster TX16S-II 
- RX hardware: Radiomaster ER6-GV
- Handset model:
- OpenTX version (including nightly number) EdgeTX 2.9.0-selfbuild
- ExpressLRS version (TX & RX MUST MATCH): TX: 3.3.0; RX: 3.3.0
- Packet Rate: 100Hz Full
- Telemetry Ratio: 1:8
- user_defines: none

pkendall64, 2023-07-27:
I think theres something else going on here.
I've just created 15 models each with a different receiver id, and model match turned on in the Lua script and these no problem whatsoever.

There is another problem that could cause the Lua lockup, especially with external modules on 500Hz, could it be that this is your problem?

analogsoftware, 2023-07-28:
PKendall64 - I think you may be right. Would this be Issue #2327 you are referring to?
I added another receiver and was able to document what is happening with screenshots.
I copied the previous model no. 15, Note Receiver ID is 0 and Status is 100Hz:
[image]
Changed to ID 15 and Status is still 100 Hz:
[image]
Changed to ID 16 and Status is now 500 Hz:
[image]
Referring to Issue #2327 changed Baudrate to 115k:
[image]
I was now able to run the Lua script. Note Packet Rate is 100Hz and Model Match ID is 0 (should be 16):
[image]
Went back to External Module setup and changed Baudrate to 400 Hz:
[image]
Still able to run the Lua script but Model Match ID is still 0:
[image]
I was distracted while working on receiver 15 and discovered the next step by accident. I power-cycled the transmitter
and when I ran the Lua script, after a delay, only the Packet Rate line displays with 500 Hz selected. For receiver 16 I timed the delay and it was 27 minutes:
[image]
I changed Packet Rate to 100hz:
[image]
After changing to 100 Hz Packet Rate the remainder of the Lua script interface appears; note the Model Match ID is now 16. After this everything works as expected.
[image]

I would not say this is a work around as waiting 27 minutes to be able to change the packet rate is not reasonable, but at least I was able to get these 2 latest receivers working. It appears there is some interaction somewhere which is setting Packet Rate to 500 Hz despite the interface showing 100 Hz.

pkendall64, 2023-07-28:
Yes, when you change the receiver number the radio sends that to the module. The module then loads it's configuration for that number, the default rate is 500Hz, which should be limited by the baud rate. This last part is where the bug is that #2327 fixes.
Maybe we just leave the issue title, but I add a fixes comment to #2327

PR #2327 (2023-07-29): External module baudrate 400k baud max air-rate is 333Hz
External modules can no longer handle 500Hz at 400k baud.
This adds an extra limit for external for modules at 400k baud to 333Hz.

Also during testing it was found that the OLED/TFT & Lua would display the configured the air-rate not the actual air-rate.
Also found was that the sync packet would have the configured air-rate and not the actual air-rate so this would cause non-connection to a receiver!

There are 2 parts to this PR to cover the 2 situations
1. Adjust the air-rate differently for internal vs external modules.
2. Save the a [... cut]
closed 2024-04-06 by pkendall64