ExpressLRS/ExpressLRS #2330: DIY 3.3 Targets

lesharodin, 2023-07-27:
Hi, could we please get DIY targets back to configurator?
Generic works fine but this confuses many users.

pkendall64, 2023-07-27:
The ones that are actually DIY are labelled as such.
Te ones that were labelled as DIY but are generic as they are what we recommend as pin configurations for manufacturers have been renamed to Generic.
This will not be changed again.

closed 2023-07-27 by pkendall64

kindssss, 2023-11-17:
hallo how to upgrade firmware 2.5.2 for 3.0.0 with target device diy_2400_esp8285_sx1280? while the target device does not exist? 
please help thx

pkendall64, 2023-11-17:
Choose these options
[image]

This question is really best asked on discord. If you have any further questions please ask in the #help-and-support channel there.