ExpressLRS/ExpressLRS #2337: Entering "other devices" crash elrs lua script.

MingchenZhang, 2023-07-30:
## Current Behavior
Current elrs LUA script would crash on entering devices menu, except the radio menu, and elrs rx menu. 
Some FC firmware provides useful tuning options in the device menu exposed in crsf protocol. Ardupilot is one example. When I switch to the ardupilot menu using "other devices" option in elrs lua menu, it crash lua script with error:
```
Syntax error
?:0
attempt to index field '?' (a nil value)
```
Elrs and ardupilot does not seems responsible since I can access those options with tbs agent lite, elrs tx and rx. 

## Steps to Reproduce
1. Use a FC with firmware that provide a device menu (if that's how it should be called) in crsf interface, such as ardupilot (at least available in arducopter and arduplane). I tested it on arducopter 4.3.0. 
2. Setup the whole elrs tx and rx thing. I tested on elrs 3.3 on both tx and rx. 
3. Execute elrs lua script. 
4. Enter "other devices"
5. select "[ArduCopter v4.3.0]"
6. Syntax error pops up. 

## Possible Solution (Not obligatory)
Use TBS agent lite to access the FC menu. 

## Details
I remember it worked before with elrs lua. That's how I find the useful menu in ardupilot. So it is likely a bug introduced in 3.X lua script. 
I wish I could help debugging, but I'm struggle to find ways to efficiently debug EdgeTX lua script. 

## Your Environment
Expresslrs v3.3.0
ArduCopter v4.3.0
- TX hardware: Built in
- RX hardware: Matek R24-D 2.4GHz RX
- Handset model: RM boxxer on Edgetx 2.8
- EdgeTX version (including nightly number)  2.8.0-factory
- ExpressLRS version (TX & RX MUST MATCH): v3.3.0
- Packet Rate: 150Hz
- Telemetry Ratio: 1:4
- user_defines: NA

deadbytefpv, 2023-08-04:
One reason it crashes when selecting Arduplane Device is that the ExpressLRS Lua Script doesn't support the Float parameter type.

Betaflight doesn't even output anything when it's selected.
Haven't tested with INAV.

See this thread in the discord for a copy of an edited lua script:
https://discord.com/channels/596350022191415318/1136939644030832661/1136961606383837214

pkendall64, 2024-04-06:
Closing as this is fixed by #2415 which will be in release 3.4.0

closed 2024-04-06 by pkendall64