ExpressLRS/ExpressLRS #2349: Radiomaster ER8GV no ability to play vario

miggy1952, 2023-08-08:
[image]
[image]
access to "play vario" in the telemetry sensor menu
No 
- TX hardware: FrSky X18S firmware 1.4.9, EU
- RX hardware: Radiomaster ER8GV running v 3.3.0
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate: 100Hz Full
- Telemetry Ratio: 1:16

miggy1952, 2023-08-08:
Thanks, I will post on the ETHOS GitHub

miggy1952, 2023-08-08:
My Matek ELRS vario is working fine, maybe it IS an ELRS issue?

miggy1952, 2023-08-08:
The working Matek one has a telemetry ID of 07 and the new not working Radiomaster has an id of 09 if that helps

pkendall64, 2023-08-08:
ELRS is calculating the altitude and vertical speed and sending the data as `CRSF_FRAMETYPE_BARO_ALTITUDE` which is type 9.

miggy1952, 2023-08-09:
OK but how does that help me to play the vario?

pkendall64, 2023-08-09:
It doesn't help you I suppose. As was mentioned before it looks like it's an ETHOS issue.

miggy1952, 2023-08-09:
[mawzthefinn](https://github.com/mawzthefinn) thought it was an ELRS issue

Hiosl, 2023-08-15:
I am having the same issue with a ER6GV, ELRS V3.3 and a Radiomaster TX16S with EdgeTX 2.7.1. Telemetry shows the altitude but not VSPEED. Was working fine before, using the Matek Vario RX.

Tried setting the ID to 009 (instead of ID007), but then it only shows altitude.

miggy1952, 2023-08-17:
Is it possible to change the sensor ID in the ER8GV? 

In ETHOS the sensor ID fields are "Greyed Out" and not available to edit.

miggy1952, 2023-08-18:
[pkendall64](https://github.com/pkendall64) tells me that.

"If we change it ETHOS then we break EdgeTX. I think EdgeTX has a much bigger community. Perhaps a bug report for ETHOS is a better place to start."

[mawzthefinn](https://github.com/mawzthefinn) told me that:

That’s wrong on 2 levels.

It won’t break EdgeTX at all. The only thing that would happen is that the RM receiver would be using the same vario sensor ID as the Matek receiver. EdgeTX understands both already, ETHOS only knows the Matek ID.

You’ve already submitted the bug report. You were asking for an interim fix until ETHOS gets updated. Changing the RM sensor ID to match the Matek one is the interim fix. It shouldn’t break anything since you aren’t going to have both active at the same time.

There’s a third aspect, is ELRS so inflexible that it can’t change sensor ID’s as a configuration item? That’s really not a good thing since it doesn’T allow for multiples of the same sensor type (which has been supported in the FrSky ecosystem since S.Port was introduced in 2013).

pkendall64, 2023-08-18:
So, my stance is it's still an ETHOS issue. The ER8GV has a barometer and we send the values as CRSF packet 9 which has both Altitude and Vertical Speed. 
This EdgeTX code show the parsing of the 0x09 packet.
https://github.com/EdgeTX/edgetx/blob/main/radio/src/telemetry/crossfire.cpp#L146-L163
And this shows the 0x07 packet.
https://github.com/EdgeTX/edgetx/blob/main/radio/src/telemetry/crossfire.cpp#L126-L128

Trying to cajole devs into making changes by saying the system is inflexible is not a way to get things moving.
By saying that ELRS is inflexible because it can't be configured to put the data on a different packet id, cuts both ways.
Is ETHOS so inflexible that it can't configure which packet it gets the data from?

closed 2023-08-18 by pkendall64

miggy1952, 2023-08-18:
Thanks for your help, looks like I should buy another Matek for my F5J fun!

I am just the guy in the middle, I believe everyone.

miggy1952, 2023-08-19:
Looks like you were correct, he now say.
OK, that's a different sort of ID than sensor ID's (which are just bus ID's and should be assignable).

They're talking about data type ID's, which do require an ETHOS update, not a simple ID change.

This is mostly a misunderstanding, sensor ID seems to mean 2 different things in the two systems.