ExpressLRS/ExpressLRS #2358: Airport serial data lost with EU_CE_2400 regulatory domain firmware

Tarick, 2023-08-14:
## Current Behavior

When either of bound devices has EU_CE_2400 firmware, during the transmit it looses symbols.

## Steps to Reproduce

I have 2 sets (pairs) of Radiomaster Ranger + Nano RP3 receiver. They have 3.3.0 release firmware and the settings Airport with 9600 bauds.
I'm using
```tio -b 9600 /dev/tty<device>```
on my linux machine where they're attached - Nano with USB-to-UART cable, Ranger with USB type C cable.
When trying to transmit from the device with EU_CE_2400 regulatory domain firmware the receiver shows the line but some symbols disappear. If both devices are in EU domain - both receivers show disappeared symbols. If only one device  - the transmitter with ISM_2400 transmits data correctly.

Since I have 2 sets with the same behavior (though identical) it is most probably firmware bug.

## Possible Solution (Not obligatory)

Using ISM_2400 domain on both devices (and sets) works flawlessly.

## Details

I'm trying to setup the serial link between my computer and the companion board on UAV. I tried every (I think) combination of the transmission frequency and baudrate without the success until I accidentally noticed that some of devices that work flawlessly are using the international regulatory domain firmware.

## Your Environment

Radiomaster Ranger  TX 2.4, Nano Divercity Receiver RP3, USB-to-UART cable, USB Type C cable.

Computer - Linux Ubuntu 23.04.

Serial connection program: tio.

pkendall64, 2023-08-16:
Airport is the base layer serial protocol over the air, it does not have retries for mossing packets. One of the devs is working on this using Mavlink with the stubborn sender in the ELRS code.

LBT shows the problem because it will not send on a channel frequency if it hears something on the channel. So it just throws the packet away that would be sent in that slot.

So this is expected behaviour. If you need reliable delivery then you need to add a retry protocol on top of the Airport. It can be at the application layer, i.e. in your code.

closed 2023-08-16 by pkendall64

Tarick, 2023-08-18:
Totally agree, but having some documentation notice about this on the Airport page would be nice, I think a lot of people will bump into this.