ExpressLRS/ExpressLRS #2363: crsf output on ch 15-16 dont work

secretman28, 2023-08-16:
## Current Behavior
on 16 ch, in sbus no problem, but in crsf output ch 15 and 16 are not working they are stable at 1811 

## Steps to Reproduce
set the protocol to 16ch and look at the output

https://www.youtube.com/watch?v=CAHO2vdnPrA

i got a kind of way to test it here the output debug from the receiver look at the 2 last bit they stay the same never change 
https://user-images.githubusercontent.com/378637/261110162-82850554-0d4b-4442-b783-720731683b37.png

im trying to bring elrs to plane guy's i made a pcb to get 16ch pwm from oxs platform like the sbus thing, i try to get elrs to the next level, all big compagnie sit on there ass for too long it is the best link that i want to make to the next level to everybody , big plane guy need this too ther servo at 333hz will be drive for the first time at the speed they are made for....
 if some one can take a look at why i cant get the ch 15 and 16 to work it will help a lot ,  from what i see its the rx that dont send the 15 and 16 on elrs but on sbus its perfect

jmxp69, 2023-08-16:
Initial issue opened with oXs https://github.com/mstrens/oXs_on_RP2040/issues/94

pkendall64, 2023-08-16:
Channel 15 and 16 are LQ and RSSI dBm.

jmxp69, 2023-08-16:
[image]
The wiki says it supports 16 full resolution channels: "16x full-resolution (10-bit, ext-limits, all half rate) with 1x 2-position channel, AUX1"

secretman28, 2023-08-16:
lq and rssi are in the telemetry why put them on a ch too?

secretman28, 2023-08-17:
you guys use 15-16 to set the dynamic ?

pkendall64, 2023-08-17:
On the receiver side the RSSI dBm and LQ are sent in the last 2 channels out the Serial connection to the flight controller so it can display them in the OSD. This is what crossfire and ghost do as well.

secretman28, 2023-08-17:
we dont use for decade this option it was to conter a problem back then , and you where setting rssi on ch something even frsky use that with the d16 , from what i see the lq rssi comming from telemetry now except if you dont have telemetry you can set it as it was on a ch , no?

pkendall64, 2023-08-17:
> we dont use for decade this option it was to conter a problem back then , and you where setting rssi on ch something even frsky use that with the d16 , from what i see the lq rssi comming from telemetry now except if you dont have telemetry you can set it as it was on a ch , no?

No, not to the flight controller. The receiver does not send telemetry to the FC, the FC only gets RC packets (or MSP packets) from the receiver and sends telemetry back to the receiver which then forwards it back to the TX module and then on to the handset.

secretman28, 2023-08-17:
mmm streamming 16ch all the time even if you use 8 not seam really optimize, ok im putting my head in to it ,  can you help me again on this like the sbus you did a killer job, many start to fall in to it ,  for test purpose is it doable to do a firmware for a betafpv nano rx that output rreal ch15-16  or its comming from the tx ?

pkendall64, 2023-08-17:
The TX only sends 16 channels if 16 channel mode is selected. If you wanted to test 16 channel CRSF, which nothing can do, you could take a look at SerialCRSF.cpp and you will see where it's putting LQ and rssidBm into the channels that are sent out the serial port.
https://github.com/ExpressLRS/ExpressLRS/blob/567bc8c4fbb574330af70f9162586a43e2bbaaca/src/src/rx-serial/SerialCRSF.cpp#L63

pkendall64, 2023-08-17:
Yes, that's correct.

secretman28, 2023-08-18:
tanks really apreciated

gjhaynes, 2023-08-18:
Hi Paul, if the TX only ends 16 channels if set to a 16-channel switch mode, would it be possible to detect this at the Rx and decide "Hey, this guy really wants to use all 16 channels, so I won't replace the received data with LQ and rssi on the serial output".

pkendall64, 2023-08-18:
Yes, that would be possible, but there are those people that want 14 channels + LQ + RSSI dBm on their FC. Especially those with a gimbal on the quad/plane.
It would perhaps be better as an option in the receiver Lua/WebUI to disable LQ/RSSI dBm output.

pkendall64, 2023-08-18:
Can you give me an actual use-case for this feature?

gjhaynes, 2023-08-18:
E.g. I have a giant scale model that uses 16 channels. It uses 2 FrSky receivers, one decoding Ch1-8 and the other Ch9-16. It would be nice to use a single ELRS receiver instead. Large turbine models also typically have a high channel count. Fixed wing pilots rely on telemetry sent back to the handset, not OSD. So it would be nice to remove the limit on 16 channel assignment by adding the config option you suggested.

secretman28, 2023-08-18:
the pwm hardware is already build and semi tested, lack of ch 15-16 stop me a bit im trying to compile the elrs with the "mod" but i just can bluid all target still compiling after 2h lolll

i want to bring elrs to the big plane community, i try my best to make it work, the sbus out help a lot for all people that have gyro and usual expander, cetafpv did a 14ch receiver to get to the plane people too i was wondering why they didint go all in on the 16ch but doing my project i find the answer hahah

pkendall64 if you wana talk come on the discord of rc video review i try to make this available to most people and he help me for that  im secretman on the discord 
 https://discord.com/channels/@me/1123412094008705155/1141918634474745956

pkendall64, 2023-08-18:
And this is for 16-channel CRSF output on the receiver serial port? What is this plugged into?

secretman28, 2023-08-18:
https://youtube.com/live/SeNsK70B4l8?feature=share

secretman28, 2023-08-18:
big plane, jet or glider use often many servo in each wing, 3 on the rudder 2 on elevetor etc... thats why big compagnie do big receiver or expander

pkendall64, 2023-08-18:
> the pwm hardware is already build and semi tested, lack of ch 15-16 stop me a bit im trying to compile the elrs with the "mod" but i just can bluid all target still compiling after 2h lolll
> 
In platformIO you should select one environment to build depending on the receiver you are using to test, either `Unified_ESP8285_2400_RX_via_UART` or `Unified_ESP32_2400_RX_via_UART`
Then it will build just the single binary!

> i want to bring elrs to the big plane community, i try my best to make it work, the sbus out help a lot for all people that have gyro and usual expander, cetafpv did a 14ch receiver to get to the [...]

No, they just ran out of pins. SBUS and PWM can output 16 channels. This is why I'm also created a target for the ESP32 S3, which has 45 GPIO pins, so we can do 16 PWM + CRSF/SBUS + dual radio + VBAT + potentially I2C for Vario/Baro.

> 
> pkendall64 if you wana talk come on the discord of rc video review i try to make this available to most people and he help me for that im secretman on the discord https://discord.com/channels/@me/1123412094008705155/1141918634474745956

That link doesn't seem to work for me and I can't find you on discord!

jmxp69, 2023-08-18:
@pkendall64 Discord invite: https://discord.gg/jSFTASU

secretman28, 2023-08-18:
> > the pwm hardware is already build and semi tested, lack of ch 15-16 stop me a bit im trying to compile the elrs with the "mod" but i just can bluid all target still compiling after 2h lolll
> 
> In platformIO you should select one environment to build depending on the receiver you are using to test, either `Unified_ESP8285_2400_RX_via_UART` or `Unified_ESP32_2400_RX_via_UART` Then it will build just the single binary!
>  tanks for that lol 

> No, they just ran out of pins. SBUS and PWM can output 16 channels. This is why I'm also created a target for the ESP32 S3, which has 45 GPIO pins, so we can do 16 PWM + CRSF/SBUS + dual radio + [...]
ok the expander i use does i2c sensor, vbat (4 different) and 16ch i probably cand add more ch but elrs was 16

> That link doesn't seem to work for me and I can't find you on discord!
 look the other post im not good at this lol

secretman28, 2023-08-18:
secretman3986 
if i understand good discord lol

jmxp69, 2023-08-18:
Thanks for working with us Paul. Appreciate it.

pkendall64, 2023-08-18:
After discussions and testing, we've decided that the behaviour should be that channel data is output on the final 2 channels.
Betaflight, Ardu and iNav have all had LinkStats packet processing for 3+ years. So since BF 4.3.0, Ardu 4.1.0 and iNav 2.6.0.
The only outlier is older DJI goggles that use the channel versions for OSD. But our stance is that they should root their goggles and use WTFOS.

PR #2365 (2023-08-19): Support 16 channel CRSF
Fixes #2363 

Link stats packet for RSSIdBm and LQ have been around for some time. So we can just send the channel data out on the channels and remove this "hack" that was there for older versions of FC software.

Betaflight 4.3.0
iNav 2.6.0
Ardu 4.1.0
closed 2023-08-19 by pkendall64