ExpressLRS/ExpressLRS #2372: No manual compilation build instructions found on site (necessary if configurator is broken for some reason)

alccrl, 2023-08-20:
Hi,
since R9M lite Pro TX Frsky external module needs st-link flashing,
but configurator 1.6.0 complains st-link not connected, (actually st-link >>IS CONNECTED<< ST IDE cube programmer connects to R9M Lite Pro),

I searched for some manual method to compile and link elrs 3.3, but I found only source code and no instructions how to use the code.

I installed platform_io, vscode and I had a look at the files and I googled in a lot of places, but I did not find any instructions.

BF, INAV, Ardupilot, EdgeTx, they do have instructions to manually compile, link and flash by st-link.
Please update documentation to give tips to people to compile elrs firmware if configurator is stuck for some reason.

I guess my specific configurator problem (no st-link recognized) is related to libusb or the like, but lack of documentation on build process is more important and urgent.

Thanks,

Alfredo

lesharodin, 2023-10-10:
https://www.expresslrs.org/software/toolchain-install/
https://www.expresslrs.org/software/stlink-fix/
Maybe it will help you

alccrl, 2023-10-10:
Thanks.
Alfredo

alccrl, 2023-11-24:
Istructions on
https://www.expresslrs.org/software/toolchain-install/
tested on Linux Ubuntu 23.10 with 22-nov-2023 master branch using a build (not flash) task.
Vscode and platformio installation had no problem.
Build result OK.

Alfredo

closed 2023-11-24 by alccrl