ExpressLRS/ExpressLRS #2373: St-link connection not recognized by configurator 1.6.0

alccrl, 2023-08-20:
Configuratore ExpressLRS
Configuratore ExpressLRS v1.6.0
Compila
Verifica della compilazione
Scarico il firmware
Compilazione firmware
Logs
Traceback (most recent call last):
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\lib\stlinkusb.py", line 130, in xfer
    self._write(cmd, tout)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\lib\stlinkusb.py", line 108, in _write

    count = self._dev.write(self._dev_type['outPipe'], data, tout)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\core.py", line 986, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\core.py", line 229, in setup_request
    self.managed_claim_interface(device, intf)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\backend\libusb0.py", line 537, in claim_interface
    _check(_lib.usb_claim_interface(dev_handle, intf))
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\usb\backend\libusb0.py", line 447, in _check

    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, invalid configuration 0\n'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "runpy.py", line 196, in _run_module_as_main
  File "runpy.py", line 86, in _run_code
  File "C:\Users\alfredo\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\ae9df3d202aa905358f0fdc1ca36718736ebdf38\firmware\flasher.pyz\__main__.py", line 3, in <module>
    if __name__ == '__main__':
  File "C:\Users\alfredo\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\ae9df3d202aa905358f0fdc1ca36718736ebdf38\firmware\flasher.pyz\_bootstrap\__init__.py", line 253, in bootstrap
  File "C:\Users\alfredo\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\ae9df3d202aa905358f0fdc1ca36718736ebdf38\firmware\flasher.pyz\_bootstrap\__init__.py", line 38, in run
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages\binary_configurator.py", line 411, in main
    return binary_flash.upload(options, args)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages\binary_flash.py", line 192, in upload

    return upload_stm32_stlink(args, options)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages\binary_flash.py", line 65, in upload_stm32_stlink
    stlink.program_flash('bootloader/' + options.bootloader, flash_start, erase=True, verify=True, initialize_comms=True)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\pystlink_api.py", line 256, in program_flash
    self.initialize_comms()
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\pystlink_api.py", line 50, in initialize_comms
    self.initialize_stlink_comms()
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\pystlink_api.py", line 64, in initialize_stlink_comms
    self.stlink = stlinkv2.Stlink(self._connector, dbg=self._dbg)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\lib\stlinkv2.py", line 90, in __init__

    self.read_version()
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\lib\stlinkv2.py", line 111, in read_version
    rx = self._connector.xfer([Stlink.STLINK_GET_VERSION, 0x80], rx_len=6, retry=2, tout=200)
  File "C:\Users\alfredo\.shiv\flasher.pyz_c4e2c0434668a3bf0810d642c43d4b4704e95dde71b23b18c3b5977eb596b6fa\site-packages/external\pystlink\lib\stlinkusb.py", line 139, in xfer
    raise stlinkex.StlinkException("USB Error: %s" % e)
pystlink.lib.stlinkex.StlinkException: USB Error: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, invalid configuration 0\n'

Risultato
Errore di compilazione
Si è verificato un errore, controlla il log per il messaggio di errore. Se non l'hai già fatto, visita Expresslrs.org e leggi la Guida Flash per il tuo particolare dispositivo e la Guida alla risoluzione dei problemi. Se hai ancora problemi dopo aver letto la documentazione, copia il log di compilazione su un sito online tipo pastebin, justpaste e sul canale #help-and-support del ExpressLRS Discord posta il link al log e altre informazioni rilevanti sul tuo dispositivo, quale metodo di flash hai usato e quali passaggi hai già fatto per risolvere il problema.

Azioni
[image]
[image]
[image]

alccrl, 2023-08-20:
PS:
Report refers to WIndows 11.

On Linux Ubuntu 23.04 it (just tested) connects and st-link  flashes Ok.

Alfredo

pkendall64, 2023-08-20:
This is a windows driver issue. We use pystlink, which uses libusb0.dll which uses the underlying windows drivers. There are a few reports available for other users of libusb which have the same or similar issues and using Zadig to install the winusb drivers has fixed the problem for those users.

alccrl, 2023-08-21:
I used also zadig, but it did not work.
May be I installed an incorrect driver from the available list.

Much probably cygwin unstallation (for arducopter sitl compilation) broke preexisting setup.

Alfredo

closed 2024-04-19 by github-actions[bot]