ExpressLRS/ExpressLRS #2382: FM30 not working on 3.3.0

CapnBry, 2023-08-25:
Something I noticed yesterday after upgrading my SIYI FM30 to ExpressLRS 3.3.0 it no longer boots (power LED doesn't even come on). Flashing 3.2.1 restores operation. This is all the information I have on it at this time.

pkendall64, 2023-08-25:
This one could be because the `-flto` option was added to `siyi.ini`. From memory it was added because we ran out of flash space, but on 3.x.x that does not seem to be the case anymore!?

CapnBry, 2023-08-26:
Any you're exactly right. Removing LTO from the build fixes it. hahaha oh boy. I believe the flash available is 61440 bytes and master compiles to 61188 bytes. I think if you hadn't done the FIFO refactor it wouldn't compile? 😅  Oop no 3.3.0 fits at 61320 bytes.

I'll look at seeing if there's some extra stuff I can ifdef out to make a little more room when PRing the change to take `-flto` back out, so FM30 can get maybe one more release until I stop carrying it.

tomerinbar1, 2024-01-02:
I'm having the same issues..
I don't know how to remove -flto from siyi.ini. Where's this file?
[image]

CapnBry, 2024-01-03:
> I'm having the same issues.. I don't know how to remove -flto from siyi.ini. Where's this file? 

Nope, you're having a different issue which is that you can't flash an FM30 from the Configurator. This needs to be fixed in the flasher but I haven't had time to work on that. I think you can just use an earlier version of the Configurator that doesn't use flasher.pyz?

tomerinbar1, 2024-01-03:
Ok I'll try to flash with an earlier version of the configurator and will let you know

tomerinbar1, 2024-01-05:
I do I know which version doesn't use flasher.pyz? or which version to download?

Thank's

closed 2024-04-19 by github-actions[bot]