ExpressLRS/ExpressLRS #2389: Jumper AION T-20 900M TX missing power settings below 250mW

DarkSoulRL, 2023-08-30:
Jumper t20.
I updated the Jumper AION T-20 900M TX internal module to version 3.3. After that, power settings below 250mW disappeared.
lua script updated to latest.

pkendall64, 2023-08-30:
It cannot go below 250mW. Based on the power output from the LoRa chip and the PA chip used.

closed 2023-08-30 by pkendall64

DarkSoulRL, 2023-08-30:
On version 2.5 there was a power of 50 and 100 MW. This confused

pkendall64, 2023-08-30:
> On version 2.5 there was a power of 50 and 100 MW. This confused

The T-20 only just came out! I think you are confusing it with the other AION module!

DarkSoulRL, 2023-08-30:
No, this is a built-in module in T20, maybe I made a mistake with version 2.5, but there were 50 and 100mW in the default firmware

pkendall64, 2023-08-30:
The 2.4 T-20 module has 25mW etc, but not the 900.
I think in the samples they provided to the devs it may have said 50mW, but was actually 250mW when measured.