ExpressLRS/ExpressLRS #2395: Ask: Increase bind mode delay from 2 seconds to 3 - Big capacitors prevent getting into bind mode

lacymorrow, 2023-09-04:
## Current Behavior

I may be wrong about this but...

Currently the bind mode (without phrase) requires power-cycling the receiver within two seconds. I have an oversized capacitor and a PDB (old equipment) that I assume are causing the rx startup to take almost 1 second to happen. Combined with the slow RX startup time, it takes over 2s for power on to occur and I cannot enter binding mode with a lipo using the 3x power-cycle method.

## Steps to Reproduce

1. Put a big capacitor on your power lead

2. Use a slow rx like the BetaFPV or HappyModel 2.4 Rx

3. Attempt the power-cycle bind method

4. No Profit.

## Possible Solution (Not obligatory)

Increase the delay/timeout for the power cycle method to 3s

## Details

## Your Environment

- TX hardware: TX16S Internal / Ranger Pro TX Module
- RX hardware: BetaFPV SuperD RX 2.4
- Handset model: 
- EdgeTX version (including nightly number): 2.9.0
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0
- Packet Rate: 50hz
- Telemetry Ratio: 1:128
- user_defines:

pkendall64, 2023-09-04:
The 2 seconds starts from when the LED first starts it's color cycling. Try that and see if it helps.

lacymorrow, 2023-09-07:
The LED takes about 2s to light up unfortunately, it could be an issue with this RX, I have two and both exhibit the issue.

pkendall64, 2023-09-07:
Yes, but the 2 seconds starts from when the LED first turns on, not from power up.

lacymorrow, 2023-09-09:
I see, interesting. I may have recieved a bad batch in that case

pkendall64, 2023-09-09:
The other way to get it into bind mode, is to hold the button on the RX down for 6+ seconds and it will clear all settings, including binding information if it was entered in the web-UI or bound via the Lua script.
Also, if you have flashed with a bind-phrase, or entered a bind-phrase on the web-UI, then power cycling 3 times will not work.

lacymorrow, 2023-09-15:
I was able to use a bind phrase with one of them, I'm almost certain the firmware is bad on the other one. I get nothing but red flashing light (supports rainbow LEDs).

closed 2023-09-15 by lacymorrow

pkendall64, 2023-09-15:
> I was able to use a bind phrase with one of them, I'm almost certain the firmware is bad on the other one. I get nothing but red flashing light (supports rainbow LEDs).

A red flashing LED right fro the start means that the firmware cannot detect the SX1280 radio chip. It should have booted into wifi mode and when you go to that you should see the hardware page with all the pin information. You could see if it has been modified or perhaps it's been flashed with the wrong hardware layout! In which case a recovery flash via the USB port and it's not fixed then theres a fault, and you should send it back.

lacymorrow, 2023-09-15:
Understood, it does, and I have re-flashed it multiple times, it occasionally shows a rainbow for one power cycle, then back to red flashing.