ExpressLRS/ExpressLRS #2397: Should modify power control while TX armed without binding?

wang322588, 2023-09-07:
## Current Behavior
I use Tlite radio.While I toggle the switch to arm with no intention and no binding.
The elrs always run on max powerlevel.

Disarmed won't make elrs module back in min powerlevel.
I think this may be wasting power with no purpose.
## Steps to Reproduce
1.Power on TX
2.No binding
3.Arm
The elrs lua setting is 250mw and dynamic power.version 3.3.0

1.

2.

3.

4.

## Possible Solution (Not obligatory)

Dypower logica code may cause this.
## Details

## Your Environment

- TX hardware:DIY F27 2.4G
- RX hardware:BetaFpv Nano rx
- Handset model:Tlite
- OpenTX version (including nightly number) edgetx lastest
- ExpressLRS version (TX & RX MUST MATCH):both 3.3.0
- Packet Rate:150Hz
- Telemetry Ratio:std
- user_defines:default setting

Meldinov, 2023-09-28:
Maybe is intentional?

pkendall64, 2023-09-29:
Yes, this is 100% by design. If you arm and do not get a telemetry back, then we jump to full (configured) power because we have no idea if this is because the receiver is out of telemetry range or not.

I'm not sure that disarming should allow the power level to go back down or not, as we are hesitant to add more things that rely on the arming state.

closed 2024-04-06 by pkendall64