ExpressLRS/ExpressLRS #2405: VTX Admin band mismatch

Drewnja, 2023-09-16:
## Current Behavior

When I try to change band with VTX admin to R, vtx changes to E, and if i try to set band F via vtx admin, it changes to R. Also power changes do not work at all, as well as pit mode. I can change these settings normally from OSD and Betaflight.

## Steps to Reproduce

1. Try to change a a band through vtx admin
2. The band on the vtx is incorrect

## Details

My vtx table is okay, because i can change bands normally from OSD and betaflight itself.

## Your Environment

- TX hardware: BETAFPV TX 868 Nano
- RX hardware: BETAFPV 868 rx
- Handset model: Jumper T-Pro (without internal elrs)
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0
- Packet Rate: 300hz
- Telemetry Ratio: Std (1:64)

lesharodin, 2023-10-10:
Can you show a VTX table from betaflight?

0crap, 2023-11-20:
Maybe related to #2452 ?

pkendall64, 2024-04-06:
Marked as closing as the requested information was not provided

closed 2024-04-19 by pkendall64