ExpressLRS/ExpressLRS #2418: Has anybody got Backpack working using ETHOS

miggy1952, 2023-09-24:
EP82 is running latest 1.3.0 backpack firmware

VTX control is working fine but I can’t get the HappyModel EP82 VRX Rapidfire Backpack to work, EP82 appears to bind OK but that is it!

On EdgeTx, everything is working fine using the same EP82 on the same Rapidfire module and using the same ELRS tx module. The only difference is if I plug in the ELRS tx into my Ethos transmitter or my EdgeTx transmitter. Backpack is working as expected with EdgeTx but not working at all when using the same components with Ethos. This made me think it might be an Lua script issue.

m0p3d, 2023-09-24:
Many things are currently broken in Ethos and they seem to be busy with their RF Module and Platform related problems. Dont expect to get a solution soon.

miggy1952, 2023-09-24:
Thanks, where is the ownership of keeping the Ethos ELRS Lua updated, is it with Frsky/Ethos or ELRS?

JyeSmith, 2023-09-24:
frsky

miggy1952, 2023-09-26:
Thanks Joe, [mawzthefinn](https://github.com/mawzthefinn) tells me that

Jye speaks for the ELRS Project, not for FrSky.

As I note above, it's up for the community to take up support of the ETHOS ELRS module script going forward. The current script is an example provided by FrSky as a starting point.

There's no obligation for the ELRS project to support this script.

miggy1952, 2023-09-26:
@bsongis-frsky
Collaborator
[bsongis-frsky](https://github.com/bsongis-frsky) commented [16 hours ago](https://github.com/ExpressLRS/ExpressLRS/issues/2418#issuecomment-1733299017)
I don't have the needed hardware to work on this.
The ELRS Lua scripts are public, they are a good start and should be easy to understand.
Don't hesitate to modify them if needed!

bsongis, 2023-09-26:
@m0p3d Would you be more specific about what is broken in Ethos? We are busy on new radios introduction, but I am not aware of something which is broken.

About Lua scripts for Ethos, I have provided many examples already and I try to answer all dev questions (last in date was Rotorflight, and I think that their script is good now)

miggy1952, 2023-10-05:
Any news/developments on this please?

JyeSmith, 2023-10-05:
Closing.  Frsky is the developer for their closed system.

closed 2023-10-05 by JyeSmith