ExpressLRS/ExpressLRS #2420: Can't build for FrSky R9m with backpack support

Unb0rn, 2023-09-27:
## Current Behavior

Flashed the build for `FrSky R9M 900MHz TX` but found out that backpack disappeared from elrs lua script, manually added the `-DUSE_TX_BACKPACK` build option but now it fails to build because it won't fit into image:

> ld: region `FLASH' overflowed by 480 bytes
> collect2: error: ld returned 1 exit status

## Steps to Reproduce

1. Set the current version (3.3.0) in the configurator

2. Set the following options:
```
-DRegulatory_Domain_EU_868
-DUNLOCK_HIGHER_POWER
-DUART_INVERTED
-DUSE_TX_BACKPACK
```
3. Build

## Details

## Your Environment

- TX hardware: FrSky R9M 2019
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0

pkendall64, 2023-09-27:
The flash space on these devices is very limited, and we've pretty much reached the end and you've found out.
We've already enabled all the optimisations we can for them too. We are at the point where we might just have to not support the STM based devices in the next major release of the firmware as well.
It might be better to start to look for some better hardware, that has longer term support. i.e. any of the ESP based hardware as they have a lot more flash space available and all of the features are enabled.

Unb0rn, 2023-09-27:
That's sad. But for now there's gotta be a way - it's "only" 480 bytes) Maybe some undocumented switches to disable these new features (like AirPort?) Also, it looks like we have LTO but at the same time we use [-O2](https://github.com/ExpressLRS/ExpressLRS/blob/435f5c97122fe792549a6bf41dac8b5f4a8ebf3b/src/targets/common.ini#L139) Shouldn't we go -Os?

DZPDZP1, 2023-12-18:
拥有相同问题，diy900 e19高频头esp32主控，lua脚本里不会出现backpack选项，编译过程会出现重新定义背包警告，但是固件可以编译成功，总之就是编译不上背包功能

DZPDZP1, 2023-12-18:
但是我发现elrs2.5版本可以成功编译上背包功能，所以希望开发者在下一个版本可以为diyesp32用户实现背包功能

closed 2024-04-06 by pkendall64