ExpressLRS/ExpressLRS #2452: VTX table is erased in BF lua script when using ELRS 3.3

AndreaPro, 2023-10-21:
## Current Behavior

I've [posted an issue](https://github.com/betaflight/betaflight-tx-lua-scripts/issues/486#issuecomment-1773490734)  on the BF lua script github and I've been suggested to report it here too. Other users reported the same issue.

## Steps to Reproduce

1. Flash ELRS 3.3 on a RX

2. Open the BF lua script and retrieve the VTX table

3. The VTX table is wrong and the night One Is overwritten on the drone

## Possible Solution (Not obligatory)

Downgrading only the RX to 3.2.1 solves the issue

## Your Environment

- TX hardware: BetaFPV Micro 1W 2,4 GHz
- RX hardware: BetaFPV ELRS AIO of BetaFPV Meteor 75 Pro
- Handset model: Radiomaster TX16S
- EdgeTX version: 2.9.0
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0
- Packet Rate: 500Hz
- Telemetry Ratio: 1:128

0crap, 2023-11-13:
Just for the record, this is 100% reproducible.
Reading goes fine, but as soon as you write the VTX settings via BF LUA the VTX table is wiped.
I stopped using it because of this and use the OSD to change VTX channels.

AndreaPro, 2023-11-13:
In [Betaflight LuaScripts GitHub ](https://github.com/betaflight/betaflight-tx-lua-scripts/issues/486)developers said that the script is working fine and they cannot fix it on their side

klutvott123, 2023-11-14:
I'm able to reproduce this with ELRS 3.3. Compiling ELRS 3.3 without msp vtx support "fixes" it. The BF lua scripts are sending a 4 byte payload for MSP_VTX_SET_CONFIG, and that should not trigger a reset of vtx tables. ELRS devs, can you please have a look at this? 🙂

amdnikos, 2023-11-21:
Same here on bf 4.4.3 edgetx 2.9.2 elrs 3.3. when i am downgrading elrs version it works as expected. Any news on this one?

jaames74, 2023-12-16:
same here, at first i thought it was something with the stm32's eeprom then i came across this thread...

jonathanoeijoeng, 2023-12-20:
I have same issue using tx radiomaster boxer elrs 3.3.0 dan rx radiomaster rp1 elrs 3.3.0 in FC Speedybee F405 v3

jaames74, 2023-12-21:
> same here, at first i thought it was something with the stm32's eeprom then i came across this thread...

RadioMaster TX16S with internal TX ELRS v3.3.0, Happymodel EP1 tco RX v3.3.0
Rolling back to v3.2.1 indeed fixes the issue.

schugabe, 2023-12-23:
I can confirm that this happens. Reloading the vtx tables is enough to trigger this issue. A fast git bisect did not yield anything useful yet

LoakAudio, 2024-01-05:
Hi all, any news for this bug ?
As BF4.5 is in testing, some of us need to use BF LUA script (easier than moving a computer on the field) and a wrong click in the script leading to loss of VTX table is very boring.
I know that rolling back to 3.2.1 fixes the thing but the goal is to have up to date versions working fine together.
Feel free to ask for test if needed

amdnikos, 2024-01-05:
Same here. It was my best feature in BF lua script. Any news from devs if it will be fixed in a next release or we have to downgrade to work only?

Στάλθηκε από Outlook για Android<https://aka.ms/AAb9ysg>

PR #2525 (2024-01-21): Disable processing MSPVTX commands if there is not SPIVTX
MSP VTX should only operate and send the VTX table if the receiver actually has a VTX.
The fix is to initialise the state machine in the STOPPED state and only enter the waiting state if there is an SPI VTX configured in the hardware layout.
Fixes #2452
closed 2024-01-21 by JyeSmith