ExpressLRS/ExpressLRS #2466: Radiomaster receiver er6gv stops working suddenly

Scachi, 2023-11-03:
I am new to all this, I am sorry for any failures doing this report and seeking for help this way...

## Current Behavior

Using a radiomaster er6gv for my fixed wing glider with 5 channels (1 channel engine, 2 channels v-tail, 2  channels ailerons)
During the configuration of the model on my table the er6gv suddenly stopped responding.
I got voice warning "rssi critical" followed by a "telemetry lost".
I waited about 2 minutes but the link wasn't established again.
The receiver linked red led slowly.
Turning the transmitter OFF and ON again didn't reestablish the connection.
Only way I could get it to reconnect was switching of the power supply of the receiver and turning the receiver back on again.

I didn't think much about it as I was during setup and it was my first elrs system.

Yesterday the same issue happened at the field flying a different fixed wing model with 3 channels (a glider type flying wing with motor to gain initial height) using a different er6gv receiver.
I was flying my glider for about 10 minutes, batteries still full, when I got the same voice warnings "RSSI critical" and "telemetry lost".
The Failsave did work and the glider came down doing slow circles 5 times.
I was about 30 meters high and 30 meters away. 
I was thinking it should establish the connection again on its own but that never happened.
With the plane on the ground I did take the receiver out while having it still connected and the led blinked red slowly.
Switching the transmitter off and on didn't reestablish the link. 
Only switching off the receiver and on again did work to establish the link again.

## Steps to Reproduce
I have no idea how to reproduce this. I happens very rarely it seems.

## Details
I was flying the two planes for several hours without any problems. It happened twice to me only.
Once during model configuration at home. Once at the field.
Two different planes, two receiver. Both radiomaster er6gv.

This is a serious problem, not being able to reestablish the link without power cycling the receiver.
To me it looks like the receiver just stops working. the led kept blinking red slowly.

## Your Environment
Transmitter:
Jumper T-Pro V2 internal ELRS
ELRS Script data:
top bar: AION T-Pro TX
Packet Rate: 100Hz Full
Telemtrie Ratio 1:16 (381bps
Switch Mode 16ch Rate/2
Model Match: Off (ID: 0)
TX Power 50mW, Dynamic Off
ELRS Version 3.3.0 CE_LBT ae9df3

EDGETX 
FW: edgetx-tprov2
Vers: "Factory (cbb2f7b1)
BUILT BY: EdgeTX
Date: 20230629

Receiver:  
RadioMaster ER6-GV 2.4GHz Diversity+6xPWM+Vario RX
Firmware Rev. 3.3.0 (ae9df3) CE_LBT

msoos, 2023-11-08:
This is a well-known issue and the ELRS developers think this is OK without cutting a new release or even notifying people on the 3.3.0 release notes. I have both extensively tested this issue and talked about it in detail and at depth in the ELRS discord. I can reproduce it all the time on the EU firmware. Bug and its fix is here: https://github.com/ExpressLRS/ExpressLRS/issues/2028 The fix got accidentally reverted for 3.3.0 (it's in 3.2.0, you can check it, if you like, it works). I have even gone so far as to test EXACTLY without and with this PR included/excluded in the firmware (yes, and I did 3+ tests each time, took many hours), and indeed, with one it disconnects all the time (in approx 18-25mins) and with the other, it never disconnects.

The PR https://github.com/ExpressLRS/ExpressLRS/pull/2438 puts this fix into the 3.x.x-maintenance However, new maintenance release is not out yet. I find this very sad but it is what it is.

If you check the ELRS discord I have raised my concern and strong displeasure that 
(1) a new release is not out and
(2) that not even a notice is added to the 3.3.0 release about this extremely serious bug

I reproduced this on another board (not RM receiver). If you lose a plane, just tell Discord, see what happens. I wanna see the face of the person who loses a 20k EUR plane due to this well-known issue that is not even in the release notes :cry:

glipski50, 2023-11-08:
@ msoos: V 3.3.1 has some LBT fixes and the update is recommended to LBT users in the release note. Have you tested with 3.3.1 as well?

msoos, 2023-11-08:
Ah, they cut a new release!!!! Yaaaaay! Sorry, homepage still says 3.3.0 is the newest, but GitHub has a new one! And indeed, the new one has the fix, it says `Fix LBT #2438` yay!

jurgelenas, 2023-11-08:
> Ah, they cut a new release!!!! Yaaaaay! Sorry, homepage still says 3.3.0 is the newest, but GitHub has a new one! And indeed, the new one has the fix, it says `Fix LBT #2438` yay!

Thanks! I have fixed the latest ExpressLRS version announcement banner in https://github.com/ExpressLRS/Docs/pull/353

closed 2023-11-20 by pkendall64